bambulab/BambuStudio #3394: Text modifier does not slice correctly when using German Umlaut Ü

elsni, 2024-01-17:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

Open attached project
Move camera so that you look on the chip from the bottom
click preview

### Actual results

One of the dots of the Ü character is missing

### Expected results

Text is complete

### Project file & Debug log uploads

[bambubug.zip](https://github.com/bambulab/BambuStudio/files/13959997/bambubug.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

elsni, 2024-01-17:
Fun fact: When I do mesh repair on the model, both Ü dots are missing.

zhimin-zeng-bambulab, 2024-01-19:
@elsni  Could you scale this model larger? Because the area of ​​the Ü dots is too small so that it cannot be extruded normally.

elsni, 2024-01-19:
I already found a workaround.. BUT:  Extruding one dot and leaving a hole where the other dot should be is unexpected behavior and should definitly be worth an investigation. Maybe the problem is the model - I was not able to reproduce it on a simple cube.
But the question is: WHY?

zhimin-zeng-bambulab, 2024-01-23:
@QingZhangBambu  Please explain why two dots of similar size, one can be extruded, and the other cannot be extruded?

QingZhangBambu, 2024-04-29:
[image]
you can adjust min line width to get wall, for the left part area just smaller than the threshold

closed 2024-06-20 by QingZhangBambu