bambulab/BambuStudio #3396: Printer Selection

FriedCheese2006, 2024-01-17:
**Is your feature request related to a problem? Please describe.**
I have two printers. One with a 0.6mm nozzle installed and one with a 0.4mm nozzle. Both have their own AMS. To switch between printers I have to:

1. Flip to the device tab and select the correct printer.
2. Wait for Bambu to connect to the printer.
3. Flip back to the prep tab and sync the AMS filaments (which updates the filament color, but not the filament profile)
[image]
4. Change the printer profile for the correct nozzle.
5. Carry on with slicing and what-not.

**Describe the solution you'd like**
Printer profiles from the prep tab that are tied to the actual printers. Have Bambu Studio generate and maintain config files for each printer (maybe tied by serial number). Let the config hold:

1. Nozzle size and type (since this is set on the printer).
2. Filament information (e.g. color and last selected profile for a given slot).

This would allow the user to just select the printer from the prep tab and everything would be updated without the flipping back and forth. Selecting a profile should also change the selected printer under the device tab so that the user is still able to access the normal settings/features there.

RyanEwen, 2024-01-17:
Funny coincidence, I just came here to suggest something very similar. 

It makes little sense to me that the Device tab is further right than the Prepare tab, since parts of the Prepare step rely on device selection (eg: to display the filaments/colors currently loaded into the AMS). Simple solution is to move the Device tab to the left the Prepare tab, but I think it can be much improved from there.

My suggestion: **Add a Device selection dropdown to the top of the Prepare sidebar under the Printer heading.** 
* When creating a new project...
  * Automatically choose the device if only one device is registered on-account. Otherwise require user to select a device.
  * Upon selecting a device, automatically select the matching Printer profile and Plate type (if possible - I assume X-series is plate-aware). 
  * Upon selecting a device, automatically update the Filament lists to include the filaments detected within the AMS (existing behavior).
* When opening an existing project...
  * Do not automatically select a device, ever, not even if the project was saved with a specific device selected. This prevents some user error and makes things more explicit / predictable.
  * Upon selecting a device, check that the selected Printer profile and Plate types match those from the project (they often won't if the project came from MakerWorld). If they do not match, ask the user if the Printer profile and Plate matching their device should be used instead, or if they want to continue with the ones from the project. 
  * Upon selecting a device, automatically update the Filament lists to include the filaments detected within the AMS (existing behavior).

walterwongbbl, 2024-01-18:
We will discuss these suggestions

MadMax389, 2024-01-18:
I would just like to have the ability to choose which printer's AMS' is/are selected from the filament tab in the slicer. This would prevent having to go to the device screen every time to get the right filaments. I don't want the AMS's to automatically sync based on the printer profile selected. I switch between 4 different printers regularly and will often print the same plate on different printers. I want the option to manually assign which slots have which filaments since my AMS's might not have the same colors in the same slots.

RyanEwen, 2024-01-18:
> I don't want the AMS's to automatically sync based on the printer profile selected.

This is something I would like to avoid as well. If it's not clear in my suggestion above, I don't want the slicer to sync the filament colors/types/profiles to what's in the AMS when the device is selected, but rather just to remember what was last used with that particular drive (instead of remember what was used in the last opened project which is almost always irrelevant to a new project, esp if the last opened project was from MakerWorld).

FriedCheese2006, 2024-01-18:
> manually assign which slots have which filaments

Isn't this available already when you hit the print button. The pop-up has dropdowns to assign which slot you want to use for a given filament in the slice. Unless I'm missing your point of course.

> remember what was last used with that particular device

That's a good point. I realize that most people probably aren't like me where I typically stick to the stock profiles.

MadMax389, 2024-01-18:
> Isn't this available already when you hit the print button. The pop-up has dropdowns to assign which slot you want to use for a given filament in the slice. Unless I'm missing your point of course.

Yes, I probably should have said that I want to keep that option as it does exist now. I guess my issue with having the printer profile tied to the physical printer (at least in my case) is that I use multiple printer profiles for the same printer. (And perhaps I'm confusing your suggestion with @RyanEwen's suggestion). For example, I have three .4mm nozzle printer profiles, each with a different minimum layer height setting for my 3 X1C's. This helps me when I use adaptive layer heights because I don't want the auto calculation to use a layer less than .16mm or .12mm. Now instead of 3 profiles, I would have 9 separate profiles (3 for each of my 3 X1C's.) That list is already cluttered enough with 4 X1C profiles an A1, an A1 Mini and a P1P. I do understand your intention and my buddy and I argue about it all the time, lol.
[image]

In my case, I don't really care which X1C I am slicing for, just that it is an X1C. One thing I did was create custom filament profiles that start with the printer model as the vendor. Once I have selected the correct printer on the device screen I can tell instantly that I'm slicing for one of the X1C's and that the AMS colors are accurate by either re-synching or selecting a filament from the dropdown.
[image]

I think everyone on the post wants the same thing, a printer selection on the prepare screen that lets you know which AMS's you are looking at so that you don't constantly have to go back and forth to the device screen. Tying that selection to a printer profile may cause more problems than it solves, in my opinion.

FriedCheese2006, 2024-01-18:
> Tying that selection to a printer profile may cause more problems than it solves, in my opinion.

Ah, I see what you mean. That's not what I meant. My thought is to have an _additional_ dropdown of some sort to select the physical printer from the prep page with the default behavior being to flip the printer profile/filament information to the last selected for that printer.

RyanEwen, 2024-01-18:
If there are multiple Printer profiles that work with the selected Device then I would prefer it not even try to auto-select or remember the last used. Remembering last used things tends to bite me whenever I make new projects as it gives the impression that things are good to go and encourages not checking. I'd rather be forced to choose from one of the compatible Printer profiles every time so that I know I am not messing up a new project just because I happened to print something earlier using a different one with different settings

FriedCheese2006, 2024-01-18:
> * Upon selecting a device, automatically select the matching Printer profile and Plate type (if possible - I assume X-series is plate-aware).

> If there are multiple Printer profiles that work with the selected Device then I would prefer it not even try to auto-select or remember the last used.

I think we need to get our story straight before the cops show up 😆.

> you are forced to choose things over again

Getting away from this is why I raised the request. It's tiresome and monotonous to have to remember every little widget that needs to be changed whenever I'm flipping from one printer to the next and often bites me (like forgetting to update the build plate type and it not being caught until the printer is halfway through it's prep cycle).

RyanEwen, 2024-01-18:
> I think we need to get our story straight before the cops show up 😆.

Story is straight - if there's _just one_ compatible Printer profile, auto select it. If there are multiple compatible Printer profiles, I prefer it not to auto select any of them (which means not remember last used either, but I imagine many will want it to select the last used for them - but not me).

MadMax389, 2024-01-18:
> and if not then I will have an empty project saved with my preferring settings, and just start from there when I want to use those settings

That is exactly what I do. I have blank template projects for each printer type. I load that first, then create a new project from the template, which keeps those settings This preserves all my standard settings for each printer type including flush volumes, plate, speeds, infill etc. If I want a new global standard setting, I save it to the template. (Actually, you can't save a blank project with no object on it, so I just added a cube primitive to save on the template project.)
[image]

TinyMito, 2024-03-04:
Here to say the same with different between with their own AMS system. I constantly have to go to device tab and switch the printer there to change the AMS between P1S Combo and A1 mini Combo.

I wish just selecting my printer profile should auto switch the connected AMS as well.