bambulab/BambuStudio #3402: Filament loading buttons cutoff

joshspires, 2024-01-18:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 23H2 Build 22635.3061

### Additional system information

Framework Laptop 13
- i5-1340P
- 16GB RAM

### Printer

Bambu Lab P1S

### How to reproduce

1. Open BambuStudio
2. Go to device tab
3. Look at the filament control section

### Actual results

The buttons overflow the width of the section, cutting a small section off. This happens regardless of the window size.
[image]

### Expected results

The buttons shouldn't go over the allowed width of the section. 

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bam-winterfell, 2024-02-21:
hi@joshspires
It is indeed a bug, and we will fix it in the future. Thank you for your feedback.

joshspires, 2024-06-15:
This looks to have been fixed in the latest release.

closed 2024-06-15 by joshspires