bambulab/BambuStudio #3404: STL file thumbnails are not displayed in Studio

AlAmantea, 2024-01-18:
### Bambu Studio Version

1.8.04.51

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Edition	Windows 11 Pro Insider Preview
Version	23H2
Installed on	‎1/‎4/‎2024
OS build	26020.1000
Experience	Windows Feature Experience Pack 1000.26020.1000.0

### Printer

X1 Carbon

### How to reproduce

Create a new project.
Import an stl file
Slice the file
print the file
do NOT save the file
exit and restart Studio

### Actual results

recent files will show the project, but not display the image.

### Expected results

Two Options:
1. Display the STL file image correctly
2. Since the file was not actually saved, ignore the file in the recent files category.

### Project file & Debug log uploads

proprietary files. no upload available.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-01-22:
@DanBao-Bambu 导入 STL 不保存，不应该在最近项目中

AlAmantea, 2024-01-22:
I agree, but it does show up there. I offer you this screen shot as proof.
[image]