bambulab/BambuStudio #3405: Add printer tabs in studio for controlling 2 or more printers.

Qjeezy, 2024-01-18:
When operating bambu studio with more than one printer, if you select a different printer from the drop down menu, all custom settings must be discarded and the new settings must be selected for the different printer or they must be transferred. Transferring doesn’t work if it’s a different printer setup and discarding the settings can be annoying when you want to go back to the first printer and have to select the proper build plate and profiles all over again. 

add tabs at the top to control more than one printer at a time by selecting the corresponding tab for which printer you’d like to control. This will also allow you to switch back within a half a second to the first printer without having to reset all the settings (profiles, build plate, and AMS filaments) to where they were. 

The only alternative to this at the moment seems to be opening a new instance of bambu studio for each printer. 

Instead of opening 4 bambu studios on my computer and slowing the computer down, make it so I can just click which printer I want to control at the top and we can switch back and forth without the slicer settings resetting or it asking for them to be transferred. Basically, allow us to have one instance of studio open and have the tabs act as if they are separate instances of studio.

tsmith35, 2024-01-19:
> When operating bambu studio with more than one printer, if you select a different printer from the drop down menu, all custom settings must be discarded and the new settings must be selected for the [...]

YES! This has been a pain since day 1. Every time I want to print something on a specific printer (say X1C #2 or P1P), I must go to the Device tab to change to the preferred device. When I go back to the prepare tab, the filaments shown belong to the *previous* device that had been selected. Now I have to do a full filament resync.

I presently have 4 printers. I can't imagine how much fun this would be with 8, 12 or more. If each printer had its own tab, the filaments loaded in the AMS could stay with whatever printer tab is selected. The constant Device->Select->Prepare->Resync mess would be resolved.

walterwongbbl, 2024-01-22:
@tsmith35 @Qjeezy 
We have received some similar requests and are currently discussing this feature.

Qjeezy, 2024-01-22:
@walterwongbbl this is great to hear! It will really streamline the Bambu Studio experience for those of us with 2 or more printers.