bambulab/BambuStudio #3406: Multi-Monitor Mode for those of us with printer farms!

amarand, 2024-01-18:
**Is your feature request related to a problem? Please describe.**
As an end user with six (going on seven) X1CC printers, I'd love to have a "multi-monitor mode" for both Bambu Studio and the Bambu Handy app. When you have six or seven jobs running on different printers, it might be nice to have it cycle through the printers one by one, sitting on them for a few seconds, then going to the next one. Better chances you'd catch spaghetti or an issue before it becomes a problem. Without this mode, it's a fairly manual process. With an automatic multi-monitor mode, I could be reading a book and just glance up, hands-free.

**Describe the solution you'd like**
I would like the application to have the ability to (this would be an option, not forced) cycle through multiple listed printers in Device mode. This includes all of the same stuff you'd see when you'd change printers in Device mode: AMS, temps, fans, and video.

**Describe alternatives you've considered**
Doing it manually, every day, all day, for over a year. Ick.

**Additional context**
Nope. Fairly simple request. Although it might be nice, when a printer is out of service for maintenance, to be able to "ignore" a printer or two, by checking/unchecking them off of a list for that cycling. Or perhaps you just ignore printers that are offline.

amarand, 2024-01-18:
Also, as one of the options for this opt-in feature, might be nice to have "number of seconds" as an option, so it can sit there for a second or two for someone who wants it to cycle rapidly, or perhaps up to a minute per system. 0-60?

raster, 2024-05-23:
The Multi-device tab in the most recent version of Bambu Studio might come close to doing what is desired, but I've found it doesn't actually update properly. If they can get it continually updating it would be great!