bambulab/BambuStudio #3407: 自定义耗材丝无法同步到AMS进行设置

iamapig120, 2024-01-18:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11 22621.3007

### Printer

Bambu Lab P1S

### How to reproduce

1. 创建自定义打印耗材，使用自定义品牌名称和自定义耗材系列名称，中英文均可复现
2. 在创建完毕后转到```设备```页面，编辑AMS耗材，找不到新创建的耗材
3. 在创建完毕后打开手机APP，编辑AMS耗材，同样找不到新创建的耗材
4. 尝试重启打印机和 Bambu Studio，仍存在此问题

### Actual results

无法在AMS耗材类型中选中自定义耗材

### Expected results

在AMS耗材类型中应可以找到并选中创建的自定义耗材

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13980547/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

QingZhangBambu, 2024-01-19:
[image]
因为一些已知的固件问题，新版studio和固件配合暂时关闭了P系列和A系列的此项功能，固件在后续版本修复后便可以正常使用

QingZhangBambu, 2024-01-19:
很抱歉给您带来的不便

iamapig120, 2024-01-19:
> 很抱歉给您带来的不便

目前基本不影响使用，但还是希望可以尽快修复，感谢拓竹各位开发者

bam-winterfell, 2024-01-19:
@iamapig120 多谢您的理解，我们会尽快修复此问题

ketoohome, 2024-03-18:
我购买的A1，遇到同样问题，至今未见修复