bambulab/BambuStudio #3408: Filament Calibration Failure

Dspinder1, 2024-01-18:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Intel(R) Core(TM) i9-9980HK CPU @ 2.40GHz   2.40 GHz, 64.0 GB (63.9 GB usable), 64-bit operating system, x64-based processor

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Go to "Calibration," Click "Flow Dynamics Calibration," or "Flow Rate Calibration.", Start the print. When it shows 100% it takes a very long time to let you continue.

### Actual results

Stops at the 100% mark and will not let me continue to record the status for a random extended period of time.

### Expected results

As soon as it reached 100%, it would let me record the status and move forward. I tried and had the same issue on my desktop computer and laptop.

### Project file & Debug log uploads

[](url)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-01-23:
@Dspinder1  Could you upload your log files? And How long did it take, a few minutes or dozens of minutes?

bust4rhymes, 2024-01-25:
[image]
I confirm i also have this issue about 3/5 times when i try to calibrate,
I'm trying to add my log file but it says that the format is not supported..!?

zhimin-zeng-bambulab, 2024-03-07:
@Dspinder1  Is your problem that the **Next** button does not appear on this page, or is the **Next** button always in a disable state?

Dspinder1, 2024-03-08:
Disabled state