bambulab/BambuStudio #3410: Constant foreground swapping

officiallymarky, 2024-01-18:
### Bambu Studio Version

1.8.4

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Ubuntu 23.10

### Printer

X1C

### How to reproduce

Print model
Minimize Bambu Studio

### Actual results

Bambu studio comes to the foreground 2-3 times at the start of a print, disrupting anything you are working on.  

### Expected results

stay minimized and just flash like a normal application unless there is a critical notification

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [X] Log file
- [ ] Project file

Qjeezy, 2024-01-18:
I tried putting in a bug report for this a long time ago but in different wording. They marked it as not a bug and told me to just sign out of studio if I didn’t want updates. The bug or problem happens when the printer updates its status (change filament, change nozzle temp, starts a different action or moves to the next gcode line). It is annoying that it doesn’t stay minimized like every other application. Fingers crossed they listen to this report and fix it.

officiallymarky, 2024-01-18:
> I tried putting in a bug report for this a long time ago but in different wording. They marked it as not a bug and told me to just sign out of studio if I didn’t want updates. The bug or problem [...]

You don't have to quit, it will do that frequently all on it's own.  Many times I minimize it, and go back to it to realize it crashed/quit on it's own.

bambu123, 2024-01-19:
@walterwongbbl 打印状态变化时，窗口会自动跑到前面，看看 linux 平台能不能发现

walterwongbbl, 2024-01-23:
https://github.com/bambulab/BambuStudio/issues/3046 @bambu123

kayvax, 2025-01-18:
I can confirm that Bambu Studio 1.10.1.50 on Linux still has this bug.
With multi filament print it is super annoying, each filament change will do that.
Bambu Studio will steal focus and open its virtual desktop disrupting whatever I am doing.

M1xel, 2025-01-28:
I experience the same issue

hadess, 2025-02-05:
Duplicate of https://github.com/bambulab/BambuStudio/issues/3046