bambulab/BambuStudio #3411: latest release makes connecting to camera much less reliable

craigyk, 2024-01-18:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 14.2.1

### Additional system information

M1 Macbook Air

### Printer

X1C

### How to reproduce

just try to connect to camera... 90% of the time just get a black image.

### Actual results

can get it to work by stopping/starting the camera multiple times until an image appears

### Expected results

should connect on the first attempt, like the previous release did

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13981593/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-01-19:
The state is playing, but showing nothing, Is that true?

craigyk, 2024-01-19:
yes, that is what it looks like… no errors or other indications of something being wrong:

￼

bambu123, 2024-01-19:
I'll check what has been changed from 1.8.3. Is that the previous version you have used?

craigyk, 2024-01-19:
Not 100% sure, but I think so.  It was the most up-to-date release prior to this recent update

bambu123, 2024-01-19:
If you don't stop it, how long would the black sceen keep on?

craigyk, 2024-01-19:
so far as long as I keep it going… it doesn’t seem to think anything is wrong, feel like the app might even be receiving video data and just not displaying it.

jmxp69, 2024-01-19:
Same issue here. Black screen on app and Bambu Studio. Usually when print stops the camera shows up again.

bambu123, 2024-01-19:
Do you have a Windows PC, Can you play and check how it performs on a Windows PC?
Is it normal to watch LIVE on a mobile phone?

bambu123, 2024-01-19:
> Same issue here. Black screen on app and Bambu Studio. Usually when print stops the camera shows up again.

Did you turn on timelapse when printing? if so, please check the timelapse video.

jmxp69, 2024-01-19:
Switching to lan only mode appears to solve it. I just formatted the card, so cannot check timelapse.

jmxp69, 2024-01-19:
> Do you have a Windows PC, Can you play and check how it performs on a Windows PC? Is it normal to watch LIVE on a mobile phone?

Failure occurs on 2 macs, windows and app.

bambu123, 2024-01-19:
> > Do you have a Windows PC, Can you play and check how it performs on a Windows PC? Is it normal to watch LIVE on a mobile phone?
> 
> Failure occurs on 2 macs, windows and app.

I think this is a printer-related issue, so I need a printer log.
could you help to create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log).

jmxp69, 2024-01-19:
I've switched to LAN mode and am in the middle of a print. When it's over, I'll switch back to online mode and see about submitting a log.

bambu123, 2024-01-19:
> I've switched to LAN mode and am in the middle of a print. When it's over, I'll switch back to online mode and see about submitting a log.

Well, thank you very much.

jmxp69, 2024-01-19:
Ticket and log submitted. 100% can replicate: LAN Only Mode camera works on 2xMac, 1xWindows and 1xiPhone App. LAN Only Mode turned off and I get black screens on all devices.

bambu123, 2024-01-19:
Ticket and log submitted. 100% can replicate: LAN Only Mode camera works on 2xMac, 1xWindows and 1xiPhone App. LAN Only Mode turned off and I get black screens on all devices.

OK, we'll check it soon.
Can you please give me the ticket No. here?

jmxp69, 2024-01-19:
US240119275001

bambu123, 2024-01-19:
> US240119275001

Are you using the A1 printer?
Then you problem is not same as @craigyk's.
Could you please also send Studio log here? (https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log)

bambu123, 2024-01-19:
> Is it normal to watch LIVE on a mobile phone?

Is it normal to watch LIVE on a mobile phone or Windows PC?

bambu123, 2024-01-19:
> so far as long as I keep it going… it doesn’t seem to think anything is wrong, feel like the app might even be receiving video data and just not displaying it.

Is it normal to watch LIVE on a mobile phone?

jmxp69, 2024-01-19:
> > US240119275001
> 
> Are you using the A1 printer? Then you problem is not same as @craigyk's. Could you please also send Studio log here? (https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log)

The problem is the same "latest release makes connecting to camera less reliable". I understand the hardware may be different, but there was a firmware update a few days ago and now "connecting to the camera is less reliable".

Studio logfile timestamps did not exactly match the printer timestamp. I uploaded Studio logs before and after the printer logfile timestamp. The files I uploaded are the closest I could find to the printer files.

jmxp69, 2024-01-19:
In LAN only mode it works.
With LAN only mode off it has issues.
Today it's working in both modes.
It's not my network connection. It's something between the camera port and the app port excluding my LAN/WAN connectivity. i.e. The issue is upstream--your cloud service somewhere.
I've removed/re-added the printer multiple times. I've rebooted the PC, Macs, and Phone. I've done all the Tier 1, Tier 2 scripted processes and have invested more time in this than I really want for a commercial product.

On the surface (without a netstat from the printer or app) the fix seems obvious: Offer local stream port for app/desktop. If printer is on lan and app can see it, allow direct connection to stream. Allow the camera local access without killing the app features. Right now if I didn't want the phone app to work, I'd have switched it to LAN only mode and been done with this. Video stream is slightly less annoying in that mode too.

craigyk, 2024-01-20:
I enjoy watching the printer perhaps a bit too much, it’s my zen place.

On another note, the iOS app seems to still show the camera reliably- but the Bambu Handy app hasn’t been updated as recently as Bambu Studio.

craigyk, 2024-01-21:
I can report that without updating my MacOS Bambu Studio, the cloud mediated video connection is working reliably again- perhaps this got fixed on the server side?

jmxp69, 2024-01-21:
> I can report that without updating my MacOS Bambu Studio, the cloud mediated video connection is working reliably again- perhaps this got fixed on the server side?

I think so too. Mine magically starting behaving yesterday. It was highly unreliable when I added my initial comment and for the last 24 hours it has been fine with no updates on my side.

jmxp69, 2024-01-21:
And today we're back to black screens. Zero changes to app/desktop/mac. And obviously my network works as I'm entering this note.

bambu123, 2024-01-22:
> And today we're back to black screens. Zero changes to app/desktop/mac. And obviously my network works as I'm entering this note.

Do you have any "tutk_log" for me ?

jmxp69, 2024-01-22:
Added to ticket

jmxp69, 2024-01-22:
After the print completed, the image showed up.

jmxp69, 2024-01-23:
In the future, you should be explicit about what logs you want. The 2nd/3rd logs from that session are no longer in the folder.  However, I'm starting a new print right now and it's doing the same thing. I had video before I sent the job to the printer, and after sending the job to the printer, the screen went black. After a short wait, I pressed the play button and saw "loading" on the right. Then loading went away, the screen remained black, and the play button turned green again. 

I will upload three logs to the ticket: 27760.

The job has been running for more than 1 minute now and I still do not have camera.

The green button has self toggled between grey/green a couple of times.

Status is sweeping x/y mech.

Status is cleaning nozzle tip.

Printing Progress is 23%

Screen remains black

I will stop the print and upload logs now.

Upon cancelling the print, the camera displayed the print table. Sometimes it does this, most of the time it does not.

jmxp69, 2024-01-23:
[image]
Uploading logs fails on your ticket system.

jmxp69, 2024-01-23:
Tried on different machine with different logs, still fails.

[image]

bambu123, 2024-01-24:
> In the future, you should be explicit about what logs you want. The 2nd/3rd logs from that session are no longer in the folder. However, I'm starting a new print right now and it's doing the same [...]

> In the future, you should be explicit about what logs you want. The 2nd/3rd logs from that session are no longer in the folder. However, I'm starting a new print right now and it's doing the same [...]

Sorry, I didn't explain it clearly. 
From your detailed description, it seems that the printer turned off LiveView during printing. In fact, the printer does turn off LiveView when starting to remotely download the model, and then opens it again after the download is complete.

I asked our colleagues on the printer side to analyze the printer logs in the ticket again.

bambu123, 2024-01-25:
> I can report that without updating my MacOS Bambu Studio, the cloud mediated video connection is working reliably again- perhaps this got fixed on the server side?

OK. Please continue to use it, and if you have any questions, please continue to report them to us.

bambu123, 2024-01-26:
@jmxp69 
Reply from our colleague on the printer side: There is a memory leak issue in the current version of A1, which may cause the inability to generate liveview images. The current suggestion is to restart the printer. The memory leak issue will be fixed in the next version and can be resolved.

We are very sorry for the inconvenience we have caused you.