bambulab/BambuStudio #3412: Bambu Studio Crashing While Slicing (on second Crash it crashes entire pc)

Vettenation, 2024-01-18:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

z790
i9-13900k 
64gb trident z 6800mah
rtx 4090

### Printer

X1C

### How to reproduce

Clicked to slice a project and one first time crashes Bambu studio, reponed and on second attempt it crashes entire pc. attached are the log files

### Actual results

Crash

### Expected results

Crash

### Project file & Debug log uploads

[crash_Thu_Jan_18_15_49_05_0_done.log](https://github.com/bambulab/BambuStudio/files/13981982/crash_Thu_Jan_18_15_49_05_0_done.log)
[crash_Thu_Jan_18_16_16_02_1_done.log](https://github.com/bambulab/BambuStudio/files/13981984/crash_Thu_Jan_18_16_16_02_1_done.log)
[crash_Thu_Jan_18_16_20_42_0_done.log](https://github.com/bambulab/BambuStudio/files/13981985/crash_Thu_Jan_18_16_20_42_0_done.log)
[crash_Thu_Jan_18_15_49_43_0_done.log](https://github.com/bambulab/BambuStudio/files/13981987/crash_Thu_Jan_18_15_49_43_0_done.log)
[crash_Thu_Jan_18_16_16_01_0_done.log](https://github.com/bambulab/BambuStudio/files/13981992/crash_Thu_Jan_18_16_16_01_0_done.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

SaltWei, 2024-01-19:
加在一下符号看看挂在哪里 @XunZhangBambu

SaltWei, 2024-01-19:
@Vettenation Could you please share the 3mf so we can try to reproduce？

Vettenation, 2024-01-19:
> @Vettenation Could you please share the 3mf so we can try to reproduce？

https://1drv.ms/u/s!AqD3xUktkYornh3V6y0TmZBjKPnt?e=ymdF6U

Kobmar96, 2024-01-22:
Same problem..

Running on i9 14900k with Z790 Dark Hero, Domintator Titanium DDR5 RAM, 4090

XunZhangBambu, 2024-01-23:
@Vettenation Can't reproduce in my machine . Both on Win10 AMD ryzen 9 5950x and win11 AMD Ryzen 5 6600H
[image]

XunZhangBambu, 2024-01-23:
@Vettenation Could you please provide the dmp file for the crash? You can refer to the following wiki link to get the dmp file.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

Vettenation, 2024-01-23:
@XunZhangBambu I attached the dump  file. Also i just tried it again and it did slice without crashing. Tried it twice. But maybe something in dump will help with what happened .

https://1drv.ms/u/s!AqD3xUktkYornnctnmnjkv2sqVAv?e=Vc6CVf

XunZhangBambu, 2024-01-24:
@Vettenation  Thanks!

Littlemanjr2, 2024-01-24:
Same issue, crashed when attempting to slice. 3mf and dmp files attached.
[PuzzleBox-2-Piece.3mf.dmp](https://github.com/bambulab/BambuStudio/files/14045005/PuzzleBox-2-Piece.3mf.dmp)
[bambu-studio.exe.3580.dmp](https://github.com/bambulab/BambuStudio/files/14045009/bambu-studio.exe.3580.dmp)

closed 2024-01-30 by Vettenation

farplain, 2024-07-01:
Was there a solution for this? I'm now experiencing the same issue, it has happened twice with two different STLs weeks apart.  Crashing the whole system. 

No dmp files being generated at all though in my case.

Kobmar96, 2024-11-13:
> > My CPU had an error in multi-core mode. I exchanged it and now it works perfectly again.
> 
> how to exchange it ? you replace cpu ?
> 
> my also i9-13900k 128gb ram rtx 4090
> 
> keep crash ... and it happen today

I did an RMA, as my CPU still was under warranty.
Do a multi-core benchmark and look if you get a bluescreen..