bambulab/BambuStudio #3415: Show/use specific names of filament from RFID data, as opposed to just a type

GuzziRaz, 2024-01-19:
I assume the filament RFID data includes a more specific name than just the type, eg. "Bambu PLA Metal". We do get a color from the RGB data but it would be way nicer if the filament name also reflected the product name (eg. "Bambu PLA Metal Iron Gray Metallic".

Calibrations should also default to the more specific name when saving.

TheIndieArmy, 2024-03-31:
I too would like to know the color names so that I can match them to my swatches. I just threw away all my boxes the filament ships in, the only place the color is printed, thinking this information would be available in Bambu Studio, but it doesn't seem to be. :(

leicazetisky, 2024-10-06:
This should be such a simple feature request to implement so not sure what's taking so long? It's especially frustrating when you have light gray, gray, dark gray and silver in your AMS and don't have the names anywhere to tell the difference when working in Bambu Studio. "upvoting" this request

hpb64, 2024-11-16:
I am red-green color blind and this would really help be because brown and green and also blue and purple look the same to me. Unless I check with my wife she ends up with green horses and blue grapes.

jfxg, 2024-12-17:
> It's especially frustrating when you have light gray, gray, dark gray and silver in your AMS and don't have the names anywhere to tell the difference when working in Bambu Studio. "upvoting" this request

This is the problem I have right now. I've been using one of the Bambu grays for a small part and now I need to order more of it but cannot figure out which specific gray it is. I'm pretty sure I've narrowed it down but it should be so simple for Bambu Studio to provide this information. I'm shocked it doesn't link you straight to the store with an "order more" button!

jonaskello, 2025-03-26:
As a side-note, you can see the full color name in the Bambu Handy android app.