bambulab/BambuStudio #3416: Text blurry in Preview window from 1.8.4 (1.8.2 is ok).

gpeden, 2024-01-19:
### Bambu Studio Version

1.8.4

### Where is the application from?

Bambu Lab github releases

### OS version

Linxux

### Additional system information

System:
  Kernel: 5.15.0-91-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: Micro-Star product: MS-7C56 v: 1.0 serial: <superuser required>
  Mobo: Micro-Star model: MPG B550 GAMING PLUS (MS-7C56) v: 1.0 serial: <superuser required>
    UEFI: American Megatrends LLC. v: 1.F0 date: 10/11/2023
Battery:
  Device-1: hidpp_battery_0 model: Logitech MX Keys Wireless Keyboard serial: <filter>
    charge: 50% (should be ignored) rechargeable: yes status: Full
CPU:
  Info: 8-core model: AMD Ryzen 7 5800X bits: 64 type: MT MCP smt: enabled arch: Zen 3 rev: 0
    cache: L1: 512 KiB L2: 4 MiB L3: 32 MiB
  Speed (MHz): avg: 2840 high: 4658 min/max: 2200/3800 boost: enabled cores: 1: 2240 2: 2240
    3: 2376 4: 4012 5: 3224 6: 2355 7: 2725 8: 3832 9: 3780 10: 4658 11: 2240 12: 2800 13: 2240
    14: 2240 15: 2240 16: 2240 bogomips: 121612
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: NVIDIA GM206 [GeForce GTX 950] vendor: PNY driver: N/A pcie: speed: 2.5 GT/s lanes: 16
    bus-ID: 2b:00.0 chip-ID: 10de:1402 class-ID: 0300
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nouveau,vesa
    unloaded: fbdev,modesetting gpu: N/A display-ID: :0 screens: 1
  Screen-1: 0 s-res: 2560x1440 s-dpi: 96 s-size: 677x381mm (26.7x15.0") s-diag: 777mm (30.6")
  Monitor-1: default res: 2560x1440 hz: 93 size: N/A
  OpenGL: renderer: llvmpipe (LLVM 15.0.7 256 bits) v: 4.5 Mesa 23.0.4-0ubuntu1~22.04.1
    direct render: Yes
Audio:
  Device-1: NVIDIA GM206 High Definition Audio vendor: PNY driver: snd_hda_intel v: kernel pcie:
    speed: 2.5 GT/s lanes: 16 bus-ID: 2b:00.1 chip-ID: 10de:0fba class-ID: 0403
  Device-2: AMD Starship/Matisse HD Audio vendor: Micro-Star MSI driver: snd_hda_intel v: kernel
    pcie: speed: 16 GT/s lanes: 16 bus-ID: 2d:00.4 chip-ID: 1022:1487 class-ID: 0403
  Sound Server-1: ALSA v: k5.15.0-91-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: Micro-Star MSI
    driver: r8169 v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: f000 bus-ID: 2a:00.0
    chip-ID: 10ec:8168 class-ID: 0200
  IF: enp42s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
Bluetooth:
  Device-1: Broadcom BCM20702A0 Bluetooth 4.0 type: USB driver: btusb v: 0.8 bus-ID: 1-6:5
    chip-ID: 0a5c:21e8 class-ID: fe01 serial: <filter>
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 2.1 lmp-v: 4.0
    sub-v: 220e hci-v: 4.0 rev: 1000
Drives:
  Local Storage: total: 1.95 TiB used: 572.84 GiB (28.7%)
  ID-1: /dev/nvme0n1 vendor: Western Digital model: WDBRPG5000ANC-WRSN size: 465.76 GiB
    speed: 31.6 Gb/s lanes: 4 type: SSD serial: <filter> rev: 111130WD temp: 52.9 C scheme: GPT
  ID-2: /dev/sda vendor: Western Digital model: WD6400AAKS-22A7B2 size: 596.17 GiB
    speed: 3.0 Gb/s type: N/A serial: <filter> rev: 3B01
  ID-3: /dev/sdb vendor: Western Digital model: WD10EZEX-00BN5A0 size: 931.51 GiB
    speed: 6.0 Gb/s type: HDD rpm: 7200 serial: <filter> rev: 1A01 scheme: MBR
Partition:
  ID-1: / size: 365.24 GiB used: 40.19 GiB (11.0%) fs: ext4 dev: /dev/nvme0n1p2
  ID-2: /boot/efi size: 190.5 MiB used: 6.1 MiB (3.2%) fs: vfat dev: /dev/nvme0n1p1
  ID-3: /home size: 915.82 GiB used: 266.33 GiB (29.1%) fs: ext4 dev: /dev/sdb1
Swap:
  ID-1: swap-1 type: partition size: 93.13 GiB used: 0 KiB (0.0%) priority: -2 dev: /dev/nvme0n1p3
USB:
  Hub-1: 1-0:1 info: Hi-speed hub with single TT ports: 10 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-2: 1-2:2 info: Genesys Logic Hub ports: 4 rev: 2.0 speed: 480 Mb/s power: 100mA
    chip-ID: 05e3:0608 class-ID: 0900
  Device-1: 1-2.3:4 info: SanDisk SDDR-B531 type: Mass Storage driver: usb-storage interfaces: 1
    rev: 2.1 speed: 480 Mb/s power: 500mA chip-ID: 0781:cfcb class-ID: 0806 serial: <filter>
  Device-2: 1-2.4:6 info: Yubicom Yubikey 4/5 OTP+U2F+CCID type: Keyboard,HID,Smart Card
    driver: hid-generic,usbhid interfaces: 3 rev: 2.0 speed: 12 Mb/s power: 30mA chip-ID: 1050:0407
    class-ID: 0b00
  Device-3: 1-5:3 info: Logitech Unifying Receiver type: Keyboard,Mouse,HID
    driver: logitech-djreceiver,usbhid interfaces: 3 rev: 2.0 speed: 12 Mb/s power: 98mA
    chip-ID: 046d:c52b class-ID: 0300
  Device-4: 1-6:5 info: Broadcom BCM20702A0 Bluetooth 4.0 type: Bluetooth driver: btusb
    interfaces: 4 rev: 2.0 speed: 12 Mb/s chip-ID: 0a5c:21e8 class-ID: fe01 serial: <filter>
  Device-5: 1-7:7 info: Micro Star MYSTIC LIGHT type: HID driver: hid-generic,usbhid
    interfaces: 1 rev: 1.1 speed: 12 Mb/s power: 500mA chip-ID: 1462:7c56 class-ID: 0300
    serial: <filter>
  Device-6: 1-10:8 info: Kensington Expert Wireless Trackball Mouse (K72359WW) type: Mouse,HID
    driver: hid-generic,usbhid interfaces: 2 rev: 1.1 speed: 12 Mb/s power: 100mA chip-ID: 047d:8018
    class-ID: 0300
  Hub-3: 2-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
  Hub-4: 3-0:1 info: Hi-speed hub with single TT ports: 4 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-5: 4-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
Sensors:
  Message: No sensor data found. Is lm-sensors configured?
Repos:
  Packages: 3455 apt: 3421 flatpak: 11 snap: 23
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/1password.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/1password-archive-keyring.gpg] https: //downloads.1password.com/linux/debian/amd64 stable main
  Active apt repos in: /etc/apt/sources.list.d/brave-browser-release.list
    1: deb [signed-by=/usr/share/keyrings/brave-browser-archive-keyring.gpg arch=amd64] https: //brave-browser-apt-release.s3.brave.com/ stable main
  Active apt repos in: /etc/apt/sources.list.d/github-cli.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/githubcli-archive-keyring.gpg] https: //cli.github.com/packages stable main
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/nordvpn.list
    1: deb https: //repo.nordvpn.com/deb/nordvpn/debian stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com virginia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  No active apt repos in: /etc/apt/sources.list.d/playonlinux.list
  No active apt repos in: /etc/apt/sources.list.d/steam-beta.list
  Active apt repos in: /etc/apt/sources.list.d/steam-stable.list
    1: deb [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
    2: deb-src [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
Info:
  Processes: 542 Uptime: 5d 8h 20m wakeups: 205 Memory: 62.72 GiB used: 20.89 GiB (33.3%)
  Init: systemd v: 249 runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12/9 Client: Cinnamon v: 6.0.4
  inxi: 3.3.13

### Printer

Bambu Lab X1C

### How to reproduce

On 1.8.4:
1. Slice plate
2. Noticed the text in the info panel is blurry

On 1.8.2:
1. Slice plate
2. Notice the text in the info plan is sharp (not blurry)

### Actual results

Info panel text is blurry

### Expected results

Info panel text should be crisp

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gpeden, 2024-01-19:
ok I installed 1.8.4 from AppImage and it looks good, so maybe its something with the new FlatPack?)

lanewei120, 2024-01-19:
hello @hadess 

any ideas?

hadess, 2024-01-19:
Do you use fractional scaling?

gpeden, 2024-01-19:
I think its a flatpak issue.  I installed the AppImage and it looks good.
a quick google says this has been a issue with other flatpack apps so might
be worth looking into that.

the workaround is to keep using the AppImage on linux.

hadess, 2024-01-20:
Can you please answer the question?

hadess, 2024-01-22:
This is a bug in BambuStudio upstream caused by https://github.com/bambulab/BambuStudio/commit/d516c0fee3f75b2e9d7cc595d1174c23f3d8c845 (merged from https://github.com/bambulab/BambuStudio/pull/3271).

Reverting it fixes the issue, and will be in Flathub in the coming hours.

hadess, 2024-01-22:
Looks like only one of the 2 commits in the PR landed in the stable v1.8.4 branch...

hadess, 2024-01-22:
2b6800b026d9f36f9ad6434aabc3962ce0236db4 is missing from the `v1.8.4` branch.

The problem didn't happen on Appimage, because it's an older version. Flatpak is at `v01.08.04.52`, appimage is only at `v01.08.04.51`.

lllucius, 2024-01-22:
> [2b6800b](https://github.com/bambulab/BambuStudio/commit/2b6800b026d9f36f9ad6434aabc3962ce0236db4) is missing from the `v1.8.4` branch.
> 
> The problem didn't happen on Appimage, because it's an older version. Flatpak is at `v01.08.04.52`, appimage is only at `v01.08.04.51`.

Sorry for the hassles.  Should I have done something differentlyi?

hadess, 2024-01-22:
> Sorry for the hassles. Should I have done something differentlyi?

Nothing, it wasn't your fault, it was the BambuStudio developers picking up only one of the commits from your PR into the stable branch, when both should have been picked up at the same time. Presumably, the person doing the merge could also have squashed both commits into one so we didn't have a broken in-between state. In either case, nothing you could have done.

lanewei120, 2024-01-23:
@LiZ-Li-BBL please check this issue
related with previous patch from @lllucius

LiZ-Li-BBL, 2024-01-23:
@hadess @lllucius 
The changes in both commits from @lllucius were merged. And the changes take effect only on Windows. This issue should not be related to the two commits.

hadess, 2024-01-23:
> The changes in both commits from @lllucius were merged. And the changes take effect only on Windows. This issue should not be related to the two commits.

It is on `master`, not in the stable branch: https://github.com/bambulab/BambuStudio/commits/v1.8.4/

lanewei120, 2024-01-23:
just missed one patch from @lllucius 
I have just pushed it into v1.8.4
and a new tag v01.08.04.53 generated

@hadess please use this one

sorry for the mistake before

hadess, 2024-01-23:
> just missed one patch from @lllucius
> I have just pushed it into v1.8.4
> and a new tag v01.08.04.53 generated

If you're going to tag a new release, please also pick up 9b47f26f20ce938a531be59742e5c1c4908ee245 which a user of the Flatpak ran into this week-end.

lanewei120, 2024-01-23:
> > just missed one patch from @lllucius
> > I have just pushed it into v1.8.4
> > and a new tag v01.08.04.53 generated
> 
> If you're going to tag a new release, please also pick up [9b47f26](https://github.com/bambulab/BambuStudio/commit/9b47f26f20ce938a531be59742e5c1c4908ee245) which a user of the Flatpak ran into this week-end.

not a new release, just for flatpak's building
does flatpak building use v1.8.4 or tag?

hadess, 2024-01-23:
> does flatpak building use v1.8.4 or tag?

We use the `v01.08.04.52` tag.

hadess, 2024-03-25:
This can be closed.

closed 2024-08-08 by lanewei120