bambulab/BambuStudio #3421: Can get the Camera on my A1 working, not the the app, or to Studio. On the same LAN as the printer

pmattss, 2024-01-19:
### Bambu Studio Version

1.8.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.2.1

### Additional system information

Every thing else works fine.  Apple Studio M1 Max, 64Gb RAM. 

### Printer

Bambu Lab A1

### How to reproduce

Click on Play in Studio for the Camera feed.  Just say Loading, nothing more, Black screen.
Same in the App. Black screen. 4

### Actual results

No Picture.

### Expected results

?

### Project file & Debug log uploads

[debug_Fri_Jan_19_12_12_54_99407.log](https://github.com/bambulab/BambuStudio/files/13988523/debug_Fri_Jan_19_12_12_54_99407.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-01-22:
The log is incomplete. Please refer to the wiki below for logs.
You should close studio first before sending log.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

bambu123, 2024-01-29:
@pmattss 
Maybe same as this https://github.com/bambulab/BambuStudio/issues/3411#issuecomment-1900972442
Could help try:
https://github.com/bambulab/BambuStudio/issues/3411#issuecomment-1911749637

pmattss, 2024-01-29:
> @pmattss
> Maybe same as this #3411 (comment)
> Could help try:
> #3411 (comment)

Sounds exactly right. Thnx.Sent from my iPad