bambulab/BambuStudio #3422: PETG-CF Calibration

AdrianPattersonMM, 2024-01-19:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1P with P1S upgrade kit installed

### How to reproduce

I am trying to calibrate my printer for PETG-CF, I have a hardened nozzle and Hardened extruder gear installed but when I click on Calibrate for either flow dynamics or flow rate I get a warning that "Printing PETG-CF material with Stainless Steel may cause nozzle damage". I clearly don't have stainless steel , so how do I get it to run the test as clicking OK just removes the error message 

### Actual results

Error Message Pops up

### Expected results

Should run calibration for the chosen filament and bed Plate

### Project file & Debug log uploads

N/A , this is not on a specific file , only on the calibration menu

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-01-22:
you can set hardened nozzle type in the printer

AdrianPattersonMM, 2024-01-22:
How do I set the hardened Nozzle type for a P1P ?

AdrianPattersonMM, 2024-01-22:
other forum posts indicate I should be able to set it here https://www.reddit.com/r/BambuLab/comments/12oq012/default_nozzle_settings_for_p1p_dont_show/

But I don't appear to have that option in my Bambu Studio 
[image]

AdrianPattersonMM, 2024-01-22:
I have managed to change the Nozzle via the bambu Handy app , 

it is odd that this cannot be done with the windows version of the software.

zhimin-zeng-bambulab, 2024-01-23:
[image]

@AdrianPattersonMM  You need to enable developer mode in "Preferences" Page

AdrianPattersonMM, 2024-01-23:
Thank you very much @zhimin-zeng-bambulab . That has fixed the issue , Hope you have a good day further

closed 2024-01-23 by AdrianPattersonMM