bambulab/BambuStudio #3423: CRITICAL: Fails on Z retractions and in the starting g-code process.

amzaldua, 2024-01-19:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.2.1

### Additional system information

Apple M1 Macbook

### Printer

Bambu Lab A1 series

### How to reproduce

Printing a G-code generated with Bambu Studio with a non planar superior surface.
I attached an example.

### Actual results

The nozzle scratch over piece.
Check the movement between 43684 and 43685, there is no a Z retraction.

### Expected results

The nozzle should be retract in Z axis between movements in the same layer and in the same type of line. Always.

### Project file & Debug log uploads

[debug_Fri_Jan_19_08_40_03_43376.log.0.zip](https://github.com/bambulab/BambuStudio/files/13989230/debug_Fri_Jan_19_08_40_03_43376.log.0.zip)
[scratching example.3mf.zip](https://github.com/bambulab/BambuStudio/files/13989232/scratching.example.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

amzaldua, 2024-01-19:
The 3D printer and the nozzle are straining, which could lead to damage to the machine. Please, fix this as soon as possible.

tsmith35, 2024-01-19:
>I think I understand the issue related to the nozzle scratching. It seems that the slicer doesn’t generate retractions when it moves rapidly within the same type of line. In other words, if the nozzle quickly moves from one XY point to another XY point, at the maximum feed rate, and both points belong to the same type of line, the slicer doesn’t retract in the Z-axis, even if the trajectory passes through lines already deposited in the same layer.

@amzaldua I saw your post and have observed the same thing. It is quite fortunate that you have "extensive knowledge of G codes and the kinematics of complex 5-axis machines" as you stated in the forum. The overwhelming majority of users would not have been able to debug as effectively at the gcode level.

Have you already added an issue for OrcaSlicer as well? Thanks.

amzaldua, 2024-01-19:
No, I haven't done it yet, but since you suggest it, I will do it to help OrcaSlicer users. Thank you.

amzaldua, 2024-01-19:
[The link to OrcaSlicer issue](https://github.com/SoftFever/OrcaSlicer/issues/3747)

tsmith35, 2024-01-19:
> @tsmith35 [The link to OrcaSlicer issue](https://github.com/SoftFever/OrcaSlicer/issues/3747)

Awesome. Thank you!

Loriborn, 2024-01-21:
Wanted to add that I've experienced this problem as well. Relatively flat prints with a lot of infill experience this the worst, and the issue seems to be exacerbated with smaller layer heights. 0.24mm seems to avoid the problem, probably due to more generous tolerances, but 0.08mm always results in the nozzle scraping over infill, regardless of the type of infill.

XunZhangBambu, 2024-01-22:
@amzaldua  Hello, personally ,this place doesn't require a Z-hop. Z-hop only occruas during retraction and layer change. And after printing, I haven't encountered a print failure. Could you please tell me why a z-hop is needed here?

[image]

tsmith35, 2024-01-22:
> @amzaldua Hello, personally ,this place doesn't require a Z-hop. Z-hop only occurs during retraction and layer change. And after printing, I haven't encountered a print failure. Could you please tell me why a z-hop is needed here?

@XunZhangBambu is it possible for users to enable a Z-hop before crossing infill? While I agree that it's not needed under ideal conditions, it's possible to drag across infill if the infill is overextruded or if the print suffers from slight warping upward.

The condition of the nozzle dragging across infill is so commonly observed that there is a nearly universal admonition on numerous Bambu-related forums to avoid grid infill whenever possible. This is troublesome for two reasons: (1) grid infill is the default infill in Studio and (2) Prusa users (and users of other printers) typically don't experience the issue. While I appreciate that skipping the Z-hop does speed up printing, it has resulted in a widespread recommendation to use gyroid infill to obtain quieter prints and avoid potential hotend damage. As you know, gyroid infill is one of the slowest infill styles available, but many users use it to avoid the grinding/scraping noise that occurs so often when using grid infill with no Z-hop over the infill.

The minimal speed gain obtained by eliminating Z-hop for increased speed across infill is a disservice to Bambu Lab's many users. That slight speed gain is strongly offset by the many, many users complaining about "loud" infill, reports of scraping noises and users thinking that their printers are being damaged by "dragging across infill" when they use traditional infill styles such as the default grid.

I can't imagine that this is the desired user experience for Bambu Lab printer users, but it's what I've observed as a regular user complaint.

SaltWei, 2024-01-22:
> is it possible for users to enable a Z-hop before crossing infill?
Disable this setting.
[image]
Modify this setting as your need
[image]

amzaldua, 2024-01-22:
I completely agree with the points raised by @tsmith35. I would like to further emphasize that another issue arising from this is the frequent occurrence of failed prints due to the piece detaching from the print bed for the same reason. Furthermore, the continuous friction between the nozzle and the piece raises concerns about potential damage to both the nozzle and the printer's mechanical components. I recommend that, by default, the option to simplify z-hop should be deactivated, and users should only enable it at their own discretion and risk.

amzaldua, 2024-01-22:
@SaltWei, by disabling "reduce infill retractions," the dragging is reduced, but only in the infill lines. The nozzle still scratches the print on the walls or in the internal solid infill.

sheikoner, 2024-01-22:
I had the same issue. I´ve got some improvement on my prints using rectilinear infill, and using z hop on my custom filament settings, and the extruder settings. But im getting the scratch nozzle issue frecuently, im not able to print more than one piece at same time because of that. Single piece often fails too, and i cannot use the AMS lite, because the purge movement its critical for this issue.

Im sure that its firmware/software problem. People who has not this issue, often prints simple things like boxes without infill, little things, and not organic models.
[image]
[image]
[image]

https://github.com/bambulab/BambuStudio/assets/116770291/02b4ac6d-8fa8-4e2e-8cb5-dae23d084857

tsmith35, 2024-01-22:
Ewww... I hate that dragging clicky sound. It sounds so nasty. What is that box of cast-away parts?

sheikoner, 2024-01-22:
> Ewww... I hate that dragging clicky sound. It sounds so nasty. What is that box of cast-away parts?

More than 2 kilos of wasted filament because of that issue. I had 95% failures. I opened a ticket with bambu lab support, but i´ve got not answer yet. 

I really think that this machine is an awesome one, but the slicer/firmware is not finished, or we have defective machines

tsmith35, 2024-01-22:
> More than 2 kilos of wasted filament because of that issue. I had 95% failures. I opened a ticket with bambu lab support, but i´ve got not answer yet.
> 
> I really think that this machine is an awesome one, but the slicer/firmware is not finished, or we have defective machines

One big thing that I discovered about 3D printing is this: no two users have the same exact situation. What works great for one person may not work at all for another person. If I had a dollar for every time a BBL customer complained about scraping / dragging / noisy infill / nozzle knocked parts loose / don't use grid infill ... well, I'd be able to buy at least a new X1C Combo with the funds. Everyone else in the world uses grid infill. Not us. It causes problems. 🙄

SaltWei, 2024-01-23:
> I would like to further emphasize that another issue arising from this is the frequent occurrence of failed prints due to the piece detaching from the print bed for the same reason

This is caused by multi-reason, most is not caused by z hop, but bad bed adhesive. You can increase the z-hop and see whether it helps. From my side, cleaning the PEI bed by water with detergent, and enabling brim does help!

SaltWei, 2024-01-23:
> @SaltWei, by disabling "reduce infill retractions," the dragging is reduced, but only in the infill lines. The nozzle still scratches the print on the walls or in the internal solid infill.

Do you means the z hop still not enough? Please show your case and 3mf, we can have a check whether the z hop is generated.

SaltWei, 2024-01-23:
> .
> ![photo_2024-01-18_12-48-29](https://private-user-images.githubusercontent.com/116770291/298753952-2afd63a6-0525-4338-8d2e-b07b75723df8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU5NzEwNDYsIm5iZiI6MTcwNTk3MDc0NiwicGF0aCI6Ii8xMTY3NzAyOTEvMjk4NzUzOTUyLTJhZmQ2M2E2LTA1MjUtNDMzOC04ZDJlLWIwN2I3NTcyM2RmOC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyM1QwMDQ1NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNzNjNmQ4MjA2Y2M5NTE0MjY3YjZiMTM5M2IzZDllMzY4YTE5M2Y3NDc2YmRiYzYyMjkwY2RiMmNlY2U1Zjc5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.6VowKbUY3rAiZwMa0X7TXLdiCvQ6Ngya3zeuy9E7Lf4)

This is not slicer issue, this is caused by bad bed bad bed adhesive. Maybe one side of the wipe tower is wrapped.

sheikoner, 2024-01-23:
No, is not. Is because nozzle hits. Do you think I will spent more than 2
kilos without leveling, calibrating flow or cleaning the plate?

Come on.

HarrisonLDavies, 2024-01-23:
Nozzle dragging happening on my P1S too, especially when ironing.

SaltWei, 2024-01-23:
> Come on.

I am not developer of printer. I can only help to check whether z hop is normal or not in your 3mf file.
Wihout 3mf file and reproduced method, I can do nothing with your cases.
Could you please share your 3mf file so we can check the z hop at the hit position? We need to debug whether it's slicer issue or other issue.
If z hop is normal, I can ask developer of printer to check further.

SaltWei, 2024-01-23:
> [Gravestone v1.3mf.zip](https://github.com/bambulab/BambuStudio/files/14017298/Gravestone.v1.3mf.zip)
> 
> Here's the 3mf file. Issues with ironing. Thanks.

Thanks, checking and printing.

tsmith35, 2024-01-23:
>This is not slicer issue, this is caused by bad bed bad bed adhesive. Maybe one side of the wipe tower is wrapped.

I think the point here is that the great majority of users *may* experience minor issues with their prints with respect to over-extrusion or non-ideal bed adhesion. Many of these issues could be easily avoided by providing z-hop whenever moving over parts of the print. Sure, perhaps the bed adhesion may not be perfect. Perhaps the infill is slightly over-extruded. Yeah, you have everyday people operating these printers. Would it hurt to give then a little z-hop? No. It would improve their user experience and give them an error-resistant printer.

Which scenario would serve Bambu Lab the best? Customers frustrated by noisy printers that make them worry that their expensive printer will become damaged, or resilient printers that give them a decent print even if they run into problems? People are not perfect, but let them have good results anyway. They will learn, and they will remember that they got the best results from Bambu Lab.

Remember Bambu Lab's ad campaign? Easy enough for your kids to use. Does that include little Jimmy crying because his printer didn't z-hop over Simba's head, instead ripping it off? Nope.

HarrisonLDavies, 2024-01-23:
> I think the point here is that the great majority of users _may_ experience minor issues with their prints with respect to over-extrusion or non-ideal bed adhesion. Many of these issues could be [...]

Indeed. My bed is clean super adhesive, belts. taught, bed level, filament dried and calibrated.

tsmith35, 2024-01-23:
> Thanks, checking and printing.

Thank you for your help.

SaltWei, 2024-01-23:
> Indeed. My bed is clean super adhesive, belts. taught, bed level, filament dried and calibrated.

Does the failure printing model still on hand? If yes, could you place it on flat surface check whether it's wrapped?

We have checked the 3mf file and gcode, the iron path and z hop of it is normal. Now we are printing it and try to reproduce. Because we don't have the filament you used, we are using Bambu PLA marble and Bambu PLA basic.

Two possible causes from my guess at the moment:

1. a little wrapping which make the top surface to be a little unflatness.

2. the iron quality is influenced by filament melting a lot. After printing big infill area and high flow extruding, when start to ironing, the melting state of filament is not good enough. Some other user met this issue before and after slowing top surface speed down before ironing, the ironing surface is with good quality. We are checking and printing this with your 3mf file as well.

SaltWei, 2024-01-23:
> > Thanks, checking and printing.
> 
> Thank you for your help.

We are checking and printing 3mf from @HarrisonLDavies . 
You can also share your cases with certain 3mf file so we can print at the same time.

amzaldua, 2024-01-23:
> > > is it possible for users to enable a Z-hop before crossing infill?
> > 
> > 
> > Disable this setting. ![image](https://private-user-images.githubusercontent.com/110660407/298478050-e18cf91d-b6f9-4a07-b536-7303e4fdadfd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU5NzEwNDYsIm5iZiI6MTcwNTk3MDc0NiwicGF0aCI6Ii8xMTA2NjA0MDcvMjk4NDc4MDUwLWUxOGNmOTFkLWI2ZjktNGEwNy1iNTM2LTczMDNlNGZkYWRmZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyM1QwMDQ1NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jODZmZDg4YzViMmM1ZWU1ZTA5YmEwMTMyOTRlZmE2MjcyZmQ5MWNiOGUyZWJlMDFkZDI0OTBjMzMyNmM5MWMwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.k6hAmhoNiCpiRgCSuyc09ScFBs8G6TC7i4bOYyBT-bQ) Modify this setting as your need ![image](https://private-user-images.githubusercontent.com/110660407/298478217-c427d00d-4da6-46b6-a40c-55acf7296c50.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU5NzEwNDYsIm5iZiI6MTcwNTk3MDc0NiwicGF0aCI6Ii8xMTA2NjA0MDcvMjk4NDc4MjE3LWM0MjdkMDBkLTRkYTYtNDZiNi1hNDBjLTU1YWNmNzI5NmM1MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyM1QwMDQ1NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04OWQyM2EyOWZkYmY5ZWFkOWQ3MDFiMmRmZDA5ZTkwZjdmZjk0NzVhODFhYTViNzM5ZjI4YzUxMzFjM2E2NGEzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.7mNrmNOTiMDc7XA_VJO_t41leMVi4N4BIDl7GKP3VUs)
> 
> @tsmith35 @amzaldua Do these settings help in your cases and solve your question about enabling a Z-hop before crossing infill?

The issue diminishes but doesn't completely disappear. The nozzle is still 'dragging' against walls, supports, etc.

SaltWei, 2024-01-23:
> The issue diminishes but doesn't completely disappear. The nozzle is still 'dragging' against walls, supports, etc.

Could you share the gcode file and picture about how the nozzle is 'dragging' so we can check it's slicer issue or other? Thanks.

HarrisonLDavies, 2024-01-23:
Hi,I have checked and it isn’t warped. Thanks. 

Harrison ***@***.***: @HarrisonDaviesInstagram: @HarrisonLDaviesBlog: ***@***.*** Author of The Er’ath Chronicles available at all good book stockists and on e-readers.CONFIDENTIAL. This message, including its attachments, if any, is intended solely for the use of the persons to whom it is addressed to and it is confidential. If you are not the intended recipient of this message, you are kindly requested to contact the sender and to delete its contents from your computer and system without reading them, taking any action based upon them, copying or disclosing them to anyone.

amzaldua, 2024-01-23:
Okay, it is very complicated for me to find free time to study the movements, but I have identified two more details that may explain the problems we are experiencing.

In the following screenshot, I marked two types of movements with arrows:

A white arrow distinguishing between arrow number one and arrow number two.
A red arrow marking a line called "wipe."
Questions:

Why is the Z-hop movement included in the same line as the repositioning in the X Y axis of the nozzle? I ask because the nozzle drags the piece even when Z-hop is activated at the start of the movement. The movements should involve first rising in the Z-axis, then moving to X Y coordinates, and finally approaching the piece in the Z-axis. Three movements, not just two.
What is the purpose of wipe lines? Probably for cleaning the nozzle, but by using these lines in the supports, it also explains our issues with dragging.
[image]

XunZhangBambu, 2024-01-23:
@amzaldua 
1.This is a feature, and we call it "slope zhop." It allows for an increase in zhop speed without causing collisions. You can choose "normal" within zhop to achieve what you mentioned

[image]

2. Wipe lines are for cleaing the nozzle.The process should be wipe (cleaning nozzle ), retract (to prevent filament leakage), travel (movement)

sheikoner, 2024-01-23:
> > Come on.
> 
> I am not developer of printer. I can only help to check whether z hop is normal or not in your 3mf file. Wihout 3mf file and reproduced method, I can do nothing with your cases. Could you please share your 3mf file so we can check the z hop at the hit position? @sheikoner We need to debug whether it's slicer issue or other issue. If z hop is normal, I can ask developer of printer to check further.

all my prints have the same issue. I cleaned the bed with soap and adding 3dlac to the surface. The nozzle ruin the print at layer 180, but starts scratching a few layers earlier.

amzaldua, 2024-01-23:
I will change this parameter and test it. Regarding the Wipe, is it not possible to disable it only in the supports? Probably, I'm going to reduce the distance or disable this feature and test it, and I'll let you know.

If these changes make the printing more reliable and smoother, don't you think you should consider defaulting to the safest possible configuration? That is Bambu Lab's philosophy—plug and play. Therefore, I understand that Bambu Studio should have a configuration that can uphold this philosophy.

By the way, thank you for your assistance.

SaltWei, 2024-01-24:
> sheikoner - Gloom Hand - GitHub test.zip

Thanks! From gcode line, I have not found abnormal point. So I find an A1 machine to print this model and check whether can reproduce.

XunZhangBambu, 2024-01-24:
> > > Come on.
> > 
> > 
> > I am not developer of printer. I can only help to check whether z hop is normal or not in your 3mf file. Wihout 3mf file and reproduced method, I can do nothing with your cases. Could you please share your 3mf file so we can check the z hop at the hit position? We need to debug whether it's slicer issue or other issue. If z hop is normal, I can ask developer of printer to check further.
> 
> [Gravestone v1.3mf.zip](https://github.com/bambulab/BambuStudio/files/14017298/Gravestone.v1.3mf.zip)
> 
> Here's the 3mf file. Issues with ironing. Thanks.
@HarrisonLDavies  Hi,after testing, your model can be successfully printed without any failures. If needed, I can provide the printed images. For ironing, you can slow down the ironing area to achieve better printing quality.

[image]
[image]

SaltWei, 2024-01-25:
> > Indeed. My bed is clean super adhesive, belts. taught, bed level, filament dried and calibrated.
> 
> Does the failure printing model still on hand? If yes, could you place it on flat surface check whether it's wrapped?
> 
> We have checked the 3mf file and gcode, the iron path and z hop of it is normal. Now we are printing it and try to reproduce. Because we don't have the filament you used, we are using Bambu PLA marble and Bambu PLA basic.
> 
> Two possible causes from my guess at the moment:
> 
> 1. a little wrapping which make the top surface to be a little unflatness.
> 2. the iron quality is influenced by filament melting a lot. After printing big infill area and high flow extruding, when start to ironing, the melting state of filament is not good enough, which cause nozzle hit and bad ironing. Some other user met this issue before, And after slowing top surface speed down before ironing, the ironing surface is with good quality. We are checking and printing this with your 3mf file as well.

@XunZhangBambu So the ironing issue from @HarrisonLDavies  should be situation 2

SaltWei, 2024-01-26:
> > sheikoner - Gloom Hand - GitHub test.zip
> 
> @sheikoner Thanks! From gcode line, I have not found abnormal point. So I find an A1 machine to print this model and check whether can reproduce.

@sheikoner I find two A1 printers, and print twice without issue. I reset all settings in your 3mf to be default and only enable tree-auto support. Because I only have bambu filament, so I change the filament to be Bambu PLA basic. 3mf/gcode files and pictures are shown below.
[file.zip](https://github.com/bambulab/BambuStudio/files/14058959/file.zip)
[image]
[image]

sheikoner, 2024-01-26:
Thanks, i tried too resetting all settings, using bambu filament, and going
down values such as my ender 3 s1 pro, and still knocking. May be hardware
or something else, because is a very common issue at A1, and P1S printers.

Thanks for checking the file. Find out a Legend of Zelda fans and gift them
my model, they will love it.

SaltWei, 2024-01-26:
> > > sheikoner - Gloom Hand - GitHub test.zip
> > 
> > 
> > @sheikoner Thanks! From gcode line, I have not found abnormal point. So I find an A1 machine to print this model and check whether can reproduce.
> 
> @sheikoner I find two A1 printers, and print twice without issue. I reset all settings in your 3mf to be default and only enable tree-auto support. Because I only have bambu filament, so I change the filament to be Bambu PLA basic. 3mf/gcode files and pictures are shown below. [file.zip](https://github.com/bambulab/BambuStudio/files/14058959/file.zip) ![img_v3_027e_a3ca144b-6112-4e2f-bc0b-f0770cc0419g](https://private-user-images.githubusercontent.com/110660407/299847728-64577e0e-20eb-4d10-b1aa-631951882056.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYyMjk5NjUsIm5iZiI6MTcwNjIyOTY2NSwicGF0aCI6Ii8xMTA2NjA0MDcvMjk5ODQ3NzI4LTY0NTc3ZTBlLTIwZWItNGQxMC1iMWFhLTYzMTk1MTg4MjA1Ni5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyNlQwMDQxMDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02YWM1YTU5MjM2YTk5MTZjOWM0NGQyMmNjNmFjMDM1YzRjMDVmMzE1OTYwZjFlNjE3NjcxZjFjOTIzYzJkOTVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.gBywLSSfZ6zjRZKNbdvhgxsNFzI0lLvrj86qEA1KVIg) ![img_v3_027f_90b0cab7-04d9-4645-a01c-7710bba0572g](https://private-user-images.githubusercontent.com/110660407/299847744-7c411fcf-fcd9-4f2c-b454-ca6459468b01.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYyMjk5NjUsIm5iZiI6MTcwNjIyOTY2NSwicGF0aCI6Ii8xMTA2NjA0MDcvMjk5ODQ3NzQ0LTdjNDExZmNmLWZjZDktNGYyYy1iNDU0LWNhNjQ1OTQ2OGIwMS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyNlQwMDQxMDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYTQzNWExNDUwNWQ2YzY1NjJjODUxZmZiMWJiYTg1NDQyYzY2YTMyMTU4ODc2Y2Y0M2M2YmJmOTMyMzc1MjIzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.bxlMh9pdCuJ9MoZlDLiCx37k8t3H5oId7s-b3C8lOps)

@sheikoner For the failure issue on this model at 180th layer, I also check the gcode in detail, but still can't find any abnormal point. From your description that the nozzle ruin the print at layer 180, but starts scratching a few layers earlier, I have two guesses.

1 Maybe something wrong with your printer, for example, the bed level does not work very well or imprecise z height, which cause bed adhesive force is not enough. The model has slight wrapping and height changing during printing
2 Different filaments have different properties. Some third party filaments are very easy to be slightly wrapping because of big internal press when printing with high speed.

For 1, Just my guess, this need to be analysed further by printer team. Could you raise a tickets by bambu handy app with more pictures and printer log, so they can continue to find the cause?

For 2,  I didn't enable timelapse recording, so you can disable as well. We can be same condition as possible and slow printing speed down and try again. Slowing down can make the filament melted well to reduce internal press which can improve the bed adhesion and reduce scratching.

SaltWei, 2024-01-26:
> Thanks for checking the file. Find out a Legend of Zelda fans and gift them my model, they will love it.

Great!

sheikoner, 2024-01-26:
> > > > sheikoner - Gloom Hand - GitHub test.zip
> > > 
> > > 
> > > @sheikoner Thanks! From gcode line, I have not found abnormal point. So I find an A1 machine to print this model and check whether can reproduce.
> > 
> > 
> > @sheikoner I find two A1 printers, and print twice without issue. I reset all settings in your 3mf to be default and only enable tree-auto support. Because I only have bambu filament, so I change the filament to be Bambu PLA basic. 3mf/gcode files and pictures are shown below. [file.zip](https://github.com/bambulab/BambuStudio/files/14058959/file.zip) ![img_v3_027e_a3ca144b-6112-4e2f-bc0b-f0770cc0419g](https://private-user-images.githubusercontent.com/110660407/299847728-64577e0e-20eb-4d10-b1aa-631951882056.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYyMjk5NjUsIm5iZiI6MTcwNjIyOTY2NSwicGF0aCI6Ii8xMTA2NjA0MDcvMjk5ODQ3NzI4LTY0NTc3ZTBlLTIwZWItNGQxMC1iMWFhLTYzMTk1MTg4MjA1Ni5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyNlQwMDQxMDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02YWM1YTU5MjM2YTk5MTZjOWM0NGQyMmNjNmFjMDM1YzRjMDVmMzE1OTYwZjFlNjE3NjcxZjFjOTIzYzJkOTVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.gBywLSSfZ6zjRZKNbdvhgxsNFzI0lLvrj86qEA1KVIg) ![img_v3_027f_90b0cab7-04d9-4645-a01c-7710bba0572g](https://private-user-images.githubusercontent.com/110660407/299847744-7c411fcf-fcd9-4f2c-b454-ca6459468b01.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYyMjk5NjUsIm5iZiI6MTcwNjIyOTY2NSwicGF0aCI6Ii8xMTA2NjA0MDcvMjk5ODQ3NzQ0LTdjNDExZmNmLWZjZDktNGYyYy1iNDU0LWNhNjQ1OTQ2OGIwMS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyNlQwMDQxMDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYTQzNWExNDUwNWQ2YzY1NjJjODUxZmZiMWJiYTg1NDQyYzY2YTMyMTU4ODc2Y2Y0M2M2YmJmOTMyMzc1MjIzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.bxlMh9pdCuJ9MoZlDLiCx37k8t3H5oId7s-b3C8lOps)
> 
> @sheikoner For the failure issue on this model at 180th layer, I also check the gcode in detail, but still can't find any abnormal point. From your description that the nozzle ruin the print at layer 180, but starts scratching a few layers earlier, I have two guesses.
> 
> 1 Maybe something wrong with your printer, for example, the bed level does not work very well or imprecise z height, which cause bed adhesive force is not enough. The model has slight wrapping and height changing during printing 2 Different filaments have different properties. Some third party filaments are very easy to be slightly wrapping because of big internal press when printing with high speed.
> 
> For 1, Just my guess, this need to be analysed further by printer team. Could you raise a ticket by bambu handy app with more pictures and printer log, so they can continue to find the cause?
> 
> For 2, I didn't enable timelapse recording, so you can disable as well. We can be same condition as possible and slow printing speed down and try again. Slowing down can make the filament melted well to reduce internal press which can improve the bed adhesion and reduce scratching.

Thanks for suggestions. Ticket is opened two or more weeks ago. The different settings on my bambu studio is because i was toching trying out to print propertly. Even i found some parameters that let me print Ok, but not as expected. Probably something like z heigh or auto calibration is causing this, because i had to relevel manually, and after that, going down with the first layer on the slicer for better addhesion.

All that things help to came out most of the prints, but i have to disable the initial bed calibration to get success prints.

Thanks for looking at the files and test it.

Best regards

SaltWei, 2024-01-26:
> Thanks for suggestions. Ticket is opened two or more weeks ago.

OK. I will ask them to handle your ticket and solve your issue ASAP!

SaltWei, 2024-01-26:
> Thanks for suggestions. Ticket is opened two or more weeks ago.

@sheikoner Hello, could you provide the ticket number? Printer team need to find your ticket raised by Bambu Handy app, and they will help to investigate further.

amzaldua, 2024-01-26:
@SaltWei I have send you the 3mf file that you request through support ticket Chanel.
Unfortunately, I cannot use BambuLab's filament to replicate the tests you perform since it is impossible for me to obtain it. I live in the Canary Islands, and for some things, it's as if I live in the middle of the Antarctic continent. However, I understand that with any other brand, I should still achieve results that, while may not have the same surface quality, won't compromise the prints.
Do you want me to print any 3MF file of your choice and send you the results? I don't have AMS Lite.

SaltWei, 2024-01-26:
> @SaltWei I have send you the 3mf file that you request through support ticket Chanel. Unfortunately, I cannot use BambuLab's filament to replicate the tests you perform since it is impossible for me [...]

@amzaldua Thank you very much! Got the file already from Customer support team. I will try to reproduce the sound you met during printing. Will let you known if need more test and information.

sheikoner, 2024-01-26:
> > Thanks for suggestions. Ticket is opened two or more weeks ago.
> 
> @sheikoner Hello, could you provide the ticket number? Is 4521989, right? Printer team need to find your ticket raised by Bambu Handy app, and they will help to investigate further.

They answered me few hours ago, thanks for move on the ticket. I provided all the info that you know to them, some 3mf files, and the link for this thread. I hope they can find out the problem to solved soon.  

Thanks again for your time and your patience.

amzaldua, 2024-02-08:
@SaltWei , the slope z-hope type needs to be corrected. As you can see on the Bambu Lab wiki, the traveling movement is missing
[image]

SaltWei, 2024-02-18:
> the traveling movement is missing

When the traveling movement is shorter than the horizontal line of slope z lift, then will directly do slope z lift.

amzaldua, 2024-02-18:
> > the traveling movement is missing
> 
> When the traveling movement is shorter than the horizontal line of slope z lift, then will directly do slope z lift and z descent without traveling. ![image](https://private-user-images.githubusercontent.com/110660407/305679027-c00607d1-e444-4f8c-bc71-8ce06390b2ad.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgyNTMyNzksIm5iZiI6MTcwODI1Mjk3OSwicGF0aCI6Ii8xMTA2NjA0MDcvMzA1Njc5MDI3LWMwMDYwN2QxLWU0NDQtNGY4Yy1iYzcxLThjZTA2MzkwYjJhZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxOFQxMDQyNTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZTJjOWMyMjk5ZDViMTA5NjgyMWQ3OGQyYTc4NTcwODA0MDE3NzAyMjJkMTc5NWEwMTBjZTUyYWI3OTk1NmE3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.c0cudXg017tDjB-KHP3JaIgTvSb2IjfGTCB00jfLj3c)

So it's not working because I've seen ramps of any length without "traveling movement." Why don't you add the ramp angle for the user to control this?

SaltWei, 2024-02-18:
> So it's not working because I've seen ramps of any length without "traveling movement.

[image]
It works on my side

SaltWei, 2024-02-18:
> Why don't you add the ramp angle for the user to control this?

The smaller the angle value, the better the anti-stringing effect. However, a flatter slope is more easy to hit object.

Currently the angle is 3 degree which can get better anti-stringing effect. When the angle is larger than 10 degree, it almost can not prevent stringing.

We can add the angle into developer mode for user to control it. This is not difficult to add it on setting page.

Meth0d007, 2024-03-13:
Coming from OrcaSlicer here for this issue as the suggested workarounds do not work for me:
https://github.com/SoftFever/OrcaSlicer/issues/3747#issuecomment-1906359833
So as i understand this is related to the z-hop now (after version 1.8 of Orca and whatever BambuStudio Version) now only doing its thing after a longer travel move then before.
Can i as a user somehow change that behavior ? My prints of large slim objects get hit of the bed by the nozzle if they are close to eachother if i slice them with Orca > 1.8

jimmyjules153, 2024-04-09:
So after reading this whole thing, the answer is there is no answer? 😥 Was this ever found to be a bug/slicer issue, or just a thing that happens? 

I get some severe scraping on my P1P. I'm trying some different settings mentioned here to see if that helps but it's failed 80% of the prints i've thrown at it. All these prints that i'm trying were working on my Creality Ender 3 V3 SE and it's super frustrating to have spent 4x the price on this P1P and to have this much difficulty with it. PLZ HALP :elmo_fire: I'll report back if i have any luck 🤷‍♀️

amzaldua, 2024-04-09:
The issue lies within the slicer and some default settings it brings.
Please configure the following in order of importance:

- Uncheck: "Quality/Others/Reduce infill retractions."
- On printer settings:
  - Set "Extruder/Retraction/Travel distance threshold/" to zero.
  - Set "Extruder/Retraction/Z hop type/" to normal or Spiral.
- Use infill patterns such as Gyroid, rectilinear, or Honeycomb.

Meth0d007, 2024-04-09:
There is more to it then these settings.

I have tested these settings multiple times and when i slice with a version > 1.8.2 (i use orcaslicer) i have plates where the nozzle crashes the model. 
It is a bit better with these settings compared to the defaults but i have large slim objects that still get knocked over in 50% of the time when sliced with orca > 1.8.2

amzaldua, 2024-04-09:
I agree, but these are just settings to alleviate the slicer issue. Let's hope that in future versions, they fix these problems because they're quite common and frustrating.

Meth0d007, 2024-04-10:
I have low hopes tbh, on Orcaslicer Repo they point the finger at BamboStudio and say it needs to be fixed here and here it does not seem to be of much interest.

XunZhangBambu, 2024-08-02:
@amzaldua 
You can check if the sticker on the nozzle  has fallen off. In our internal use, the sticker falling off may also be the source of the scraping sound.

https://github.com/user-attachments/assets/a5546d20-85c1-4660-ab0c-9949013bc3f4

XunZhangBambu, 2024-08-02:
@amzaldua You can also try the infill pattern"Cross Hatch" which can significantly reduce scraping in infill
[image]

[85 more replies]