bambulab/BambuStudio #3426: CRITICAL: The extruder pushes the filament while it is cold.

amzaldua, 2024-01-19:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Sonoma 14.2.1

### Additional system information

Apple M1 Macbook

### Printer

Bambu Lab A1

### How to reproduce

Start any g-code program sliced by BambuStudio.

### Actual results

During the initial movements, the extruder forces the filament before it reaches the fusion temperature, straining the mechanisms and the motor.

### Expected results

The extruder should not activate until the filament has reached the correct temperature.

### Project file & Debug log uploads

[Esfera_PLA_40m22s.gcode.zip](https://github.com/bambulab/BambuStudio/files/13989359/Esfera_PLA_40m22s.gcode.zip)
[debug_Fri_Jan_19_08_40_03_43376.log.0.zip](https://github.com/bambulab/BambuStudio/files/13989360/debug_Fri_Jan_19_08_40_03_43376.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

amzaldua, 2024-01-19:
I have a ticket created with the number US240112563001 since 2024/01/12 without any response.

tsmith35, 2024-01-19:
>I think I understand the issue related to the nozzle scratching. It seems that the slicer doesn’t generate retractions when it moves rapidly within the same type of line. In other words, if the nozzle quickly moves from one XY point to another XY point, at the maximum feed rate, and both points belong to the same type of line, the slicer doesn’t retract in the Z-axis, even if the trajectory passes through lines already deposited in the same layer.

@amzaldua I saw your post and have observed the same thing. It is quite fortunate that you have "extensive knowledge of G codes and the kinematics of complex 5-axis machines" as you stated in the forum. The overwhelming majority of users would not have been able to debug as effectively at the gcode level.

Have you already added an issue for OrcaSlicer as well? Thanks.

amzaldua, 2024-01-19:
The printer tries to push the filament automatically at the beginning of each printing process. I can only watch helplessly as the extruder creaks.

amzaldua, 2024-01-23:
Hello!
Any progress on this issue?

pichen2021, 2024-01-24:
Hi @amzaldua ,
This is a necessary action and will not cause abnormality. If you need to remove this action, you can delete this part of gcode in start gcode:
[image]

amzaldua, 2024-01-24:
> Hi @amzaldua , This is a necessary action and will not cause abnormality. If you need to remove this action, you can delete this part of gcode in start gcode: [...]

Is it necessary to force the extruder? Why? The sound that the extruder produces doesn't indicate that to me.
Could you confirm that, @XunZhangBambu? If this is the case, I would like an official confirmation that this is normal and does not pose any risks to the integrity of the extruder, neither in the short nor long term.

facuxzsummoner, 2024-04-14:
Finally solved this issue!

At least for me this was my solution: 

https://wiki.bambulab.com/en/a1-mini/maintenance/manual-bed-tramming

the bed isn't correctly leveled from the factory.
After doing the manual tramming, recalibrated the printer and tryed a print that always hit the nozzle for me. it finnly works!!!!

Flexigem, 2024-07-16:
Are you sure your answer is in the right thread? Your issue does not correlate to this issue.

Flexigem, 2024-07-16:
> > Hi @amzaldua , This is a necessary action and will not cause abnormality. If you need to remove this action, you can delete this part of gcode in start gcode: ![image](https://private-user-images.githubusercontent.com/141391188/299280616-a924bd79-fa02-4687-953b-649ecc1c0659.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYwOTUzMDksIm5iZiI6MTcwNjA5NTAwOSwicGF0aCI6Ii8xNDEzOTExODgvMjk5MjgwNjE2LWE5MjRiZDc5LWZhMDItNDY4Ny05NTNiLTY0OWVjYzFjMDY1OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyNFQxMTE2NDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03OTc0OGM2ZjE3ZmM3MjQ4ZTUyMmVjMmRmYjEyNmNhMGM2MDQ4MzA3N2MzYmZiZjc5YzVmMDAzYTExYWFhMTA3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.g9AsrtOP_C_sTNkqPuKHl-cOOcZLb_L1giTItvFyXhk)
> 
> Is it necessary to force the extruder? Why? The sound that the extruder produces doesn't indicate that to me. Could you confirm that, @XunZhangBambu? If this is the case, I would like an official confirmation that this is normal and does not pose any risks to the integrity of the extruder, neither in the short nor long term (no offense intended, @pichen2021 )

@Bambulab
How about making an official statement about this weird procedure which no other Printer start code does have this.

gaolegao-lx, 2024-07-17:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Flexigem, 2024-07-19:
> **Please upload log through Handy APP, and give us your ticket number**
>
> _**Upload log files through Bambu Handy**_ Log files from the Bambu printer can be uploaded via Bambu Handy over the network, [...]

here you are:
US240719351002

meanwhile I removed this extrusion code as it does not make sense as long as there is not an official statement why it is there. The A1 mini does not have that code in the start Gcode
[image]