bambulab/BambuStudio #3427: Change Colors and Forms

daluro, 2024-01-19:
### Bambu Studio Version

v01.08.04.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Series

### How to reproduce

[image]

If you click on "Höhenreichweite" and do some Changes, you are dont 
you will then no longer be able to make other settings. you can only change the colours via "Numbers". you cannot do anything with the mouse. you can click but nothing happens

### Actual results

nothing happens

### Expected results

you can change the tools

### Project file & Debug log uploads

/nV

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-01-22:
[image]
Hello, can you provide some logs according to this operation according this gif. If you could record a video of this issue, it would be even more convenient for us to solve this problem. thanks

Haidiye00, 2024-05-14:
Please download and use the latest version. If there are still any issues, please use pictures or screenshots to describe your ideas. Thank you

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00