bambulab/BambuStudio #3428: Error after Update on the PC running Windows 11--Fatal Error, exception catches invalid string position

Craigphi, 2024-01-19:
### Bambu Studio Version

v01.01.04.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X!C

### How to reproduce

I downloaded the new update.  I cannot open Bambu Studio. I get the error message: Fatal Error, exception catches invalid string position. I have installed it a second time and get the same error. So I cannot open Bambu Studio now after the update. 

### Actual results

Fatal Error, exception catches invalid string position.

### Expected results

I expect Bambu Stuiio to open

### Project file & Debug log uploads

No file to attach

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-01-22:
this is very likely caused by the config file corrupted
could you give a snapshot of folder?
[image]

and please try following steps:
1. close bambustudio
2. remove C:\Users\user.name\AppData\Roaming\BambuStudio\system
and try again
if still fails
3.  try to backup and then remove C:\Users\user.name\AppData\Roaming\BambuStudio\BambuStudio.conf
then try again

Craigphi, 2024-01-23:
Thank you for your reply. I solved the problem by uninstalling the app via Windows add-remove programs and deleting all user files. Then I reinstalled BambuStudio and it works again. Thanks again,Craig Phillips Sent from my iPhone

image.png (view on web)

close bambustudio

lanewei120, 2024-01-24:
Glad to see this issue solved

closed 2024-01-24 by lanewei120

MangoPies, 2025-03-29:
Thank you for the help. Had the same issue. The last step to delete bambustudio.conf was the fix

GetUpdate, 2025-06-25:
Thank you deleting the .conf solved it ! 
Whish you all the best 
CyberHome