bambulab/BambuStudio #3430: Automatic scaling of Parts (For Shrinkage)

brisawaldbauer, 2024-01-19:
**Is your feature request related to a problem? Please describe.**
Every time I print in ABS I need to scale my part by a certain percentage in order to account for shrinkage. It is time consuming to manually change each part and I also often forget, leading to parts that don't fit together which creates a ton of waste as I'm forced to completely reprint pieces that I forgot to scale.

**Describe the solution you'd like**
It would be amazing if the Filament Settings window had a tab called Part Scaling where you could enter values that you want to scale you parts by in x, y, and z every time they are to be printed using that specific filament.  For example, in the filament settings for ABS I could set the scaling values to 100.62% in x and y. Then every time I import a model into Bambu Studio and set it to be printed in ABS, it would automatically get scaled by 100.62% in x & y. I think it would also be important to have the applied scaling shown in the Scale dropdown (instead of it showing 100%, it would show 100.62%) so that you never accidentally double scale the piece.

**Describe alternatives you've considered**
The only clear alternative is to scale each part manually using the Scale dropdown while the part is selected.

**Additional context**

amzaldua, 2024-01-20:
Yes, please. At filament level like Orca Slicer.

Qjeezy, 2024-01-22:
I agree. Orcaslicer has “shrinkage” in the filament profile that can be adjusted to do exactly what you want here. Bambu should import this feature to Studio from OrcaSlicer.

stony2k, 2024-02-03:
I agree. My X1 prints PLA with exact 0.2 mm undersize. When scaling the part i. X and Y to 100.12 % the real part ist about 99,99% to the real measures, which it good enough. Some shrinkage option is much appreciated instead of scaling every part every time.