bambulab/BambuStudio #3431: Printing multiple items as one object

Mikebau5, 2024-01-19:
For printing multiple small items in different colors there is no option to print multiple objects as 'one' object before switching filament.

I'd like to see an option to select items to print per layer (let's say the 8 ducks) before switching to the blue card holder items (and print them together per layer as if it were one object). This helps to prevent a huge filament waste

[image]

[image]

MadMax389, 2024-01-22:
Does it work if you assemble all the ducks into one object?

Mikebau5, 2024-01-22:
> Does it work if you assemble all the ducks into one object?

How does one achieve this exactly. I couldn't find this option in Bambu Studio, either that or I'm blind

MadMax389, 2024-01-22:
Select them all, r-click, assemble.

Mikebau5, 2024-01-22:
Okay Epic, that worked exactly as I hoped!

closed 2024-01-22 by Mikebau5