bambulab/BambuStudio #3432: Bambu Studio Crashes when trying to view camera

etazelaar, 2024-01-19:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 & Ubuntu 22.04 Flatpak

### Additional system information

When I try to view the camera on either my X1 Carbons with the latest 1.8.4.51 Bambu Studio locks up and becomes unresponsive. It also happens sometimes on large slices.

### Printer

Bambu X1 Carbon

### How to reproduce

Click on the "Play" button to view the camera.

### Actual results

The program freezes and must be stopped by the OS.

### Expected results

View built-in cameras.

### Project file & Debug log uploads

not available

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-01-22:
Please refer this wiki
https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view

fahaaaxiki, 2024-12-06:
Hi， we have solved this issue in our latest Public version：[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50), you can update to this version to check if it works