bambulab/BambuStudio #3436: Mac P1S printer video feed issues

SyntaxError6, 2024-01-19:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OS 14.2.1

### Additional system information

2.4 GHz 8-Core Intel Core i9 AMD Radeon Pro 5300M 4 GB  Intel UHD Graphics 630 1536 MB 16 GB 2667 MHz DDR4

### Printer

Bambulab P1S

### How to reproduce

Watch live video stream on MAC while printing. 

iOS app on same network works just fine. 

### Actual results

Video swaps between live video and blue video with green vertical lines. Sometimes instead of the green lines video it will be a still frame from 20-30 seconds in the past.

[image]

### Expected results

Doesn't do this

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13995479/log.zip)
[DryPods_FullSet_wDessicant+Trays.3mf.zip](https://github.com/bambulab/BambuStudio/files/13995485/DryPods_FullSet_wDessicant%2BTrays.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-01-22:
As a contrast, could you help try LAN Only Mode camera?

bambu123, 2024-01-25:
@SyntaxError6 
Is there a normal playback time, such as when the printer is not printing?
Do you have any other MacBook? Is the liveview on it normal?

SyntaxError6, 2024-01-27:
Hello @bambu123 

Sorry it took so long to respond. 

I have tried another Macbook Pro and it does the same thing. I also have a windows machine on the same network and it has no issues like this. 

I haven't tried the LAN Only Mode yet. Can you provide a link with instructions on how to do that on my P1S

Thanks

bambu123, 2024-01-29:
> Hello @bambu123
>
> Sorry it took so long to respond.
>
> I have tried another Macbook Pro and it does the same thing. I also have a windows machine on the same network and it has no issues like this.
>
> I [...]

Please help try LAN Only mode. 
Set LAN Only on the printer, then find the LAN mode printer on Studio and bind it.
Then you can play liveview.

SyntaxError6, 2024-01-29:
Ok set it to LAN only mode and I still see the original issue. I attached a video of bambulabs with the issue. 

https://github.com/bambulab/BambuStudio/assets/26724633/fcf60106-7896-4049-90ad-26cc072c73e7

SyntaxError6, 2024-01-30:
Here you go. 

[Terminal Saved Output.txt](https://github.com/bambulab/BambuStudio/files/14102679/Terminal.Saved.Output.txt)

bambu123, 2024-01-31:
> Here you go. I did it twice thinking I had done it incorrectly.
> 
> [log.zip](https://github.com/bambulab/BambuStudio/files/14102711/log.zip) [Terminal Saved Output.txt](https://github.com/bambulab/BambuStudio/files/14102679/Terminal.Saved.Output.txt)

Thank you!

bambu123, 2024-02-01:
@SyntaxError6 
These days I found several points that may cause this problem and made some modifications. Can you help me test and collect logs?
I found that there may be a binary signature issue, although I have signed the file. So I added a step (2).

How TO:
1、Replace libBambuSource.dylib with the following， it is located at ~/Library/Application\ Support/BambuStudio/plugins
2、Open  libBambuSource.dylib by context menu, click OK on the warning dialog.
3、Open BambuStudio from terminal, /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
4、Play liveview, reproduce the problem, save the log from terminal
5、Send back the terminal log together with Studio logs

SyntaxError6, 2024-02-03:
Sorry for the delay. Was mid project and needed to get something done. I have provided the logs and video capturing the green bars as well as the console window so you can synchronize the logs with the event. 

[Terminal Saved Output.txt](https://github.com/bambulab/BambuStudio/files/14152996/Terminal.Saved.Output.txt)
[log.zip](https://github.com/bambulab/BambuStudio/files/14153120/log.zip)

https://github.com/bambulab/BambuStudio/assets/26724633/9d3ca147-b817-4549-8992-136fd8816252

bambu123, 2024-02-04:
@SyntaxError6 
Can you help me try the second modified version again?
[libBambuSource.dylib.zip](https://github.com/bambulab/BambuStudio/files/14154733/libBambuSource.dylib.zip)

SyntaxError6, 2024-02-04:
That second zip file seems corrupt. When I try to unzip it I g
[image]
et an error.

bambu123, 2024-02-05:
> That second zip file seems corrupt. When I try to unzip it I get an error. ![Screenshot 2024-02-04 at 10 46 05 AM](https://private-user-images.githubusercontent.com/26724633/302114677-4b8fa4f4-df0f-4748-ab27-7c46307180b7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcwNjE4ODksIm5iZiI6MTcwNzA2MTU4OSwicGF0aCI6Ii8yNjcyNDYzMy8zMDIxMTQ2NzctNGI4ZmE0ZjQtZGYwZi00NzQ4LWFiMjctN2M0NjMwNzE4MGI3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA0VDE1NDYyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWExYTQ0YTAxNjVkOGE2YTQ0ZmNjMjQ4MmIwYWE5MzAzZjcxNzQwNjBmNDM5MjY3NGI5Y2JlOWM5YmE2YTA2M2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.kjmkRXvLRWZLn8G6zG_vOOuHFa4a714Bn29r8e93MQQ)

Please try this again
[libBambuSource 2.dylib.zip](https://github.com/bambulab/BambuStudio/files/14159300/libBambuSource.2.dylib.zip)

SyntaxError6, 2024-02-05:
@bambu123 I think that fixed it. I don't see the green bars and the video updates are not going back in time. 🎉 

[Terminal Saved Output.txt](https://github.com/bambulab/BambuStudio/files/14166785/Terminal.Saved.Output.txt)
[log.zip](https://github.com/bambulab/BambuStudio/files/14166797/log.zip)

bambu123, 2024-02-06:
> @bambu123 I think that fixed it. I don't see the green bars and the video updates are not going back in time. 🎉
> 
> [Terminal Saved Output.txt](https://github.com/bambulab/BambuStudio/files/14166785/Terminal.Saved.Output.txt) [log.zip](https://github.com/bambulab/BambuStudio/files/14166797/log.zip)

Thanks for your report. I will release a plugin update recently.

SyntaxError6, 2024-02-06:
No problem. Happy to help. 

Really enjoying my P1S I just purchased. Thank you for helping to resolve this issue. If there is anything else I can help with around issues with the mac build let me know. 

Nick 

My Profile:
[https://makerworld.com/en/u/1429899458](url)

bambu123, 2024-02-06:
Resolved, please follow the prompts on Studio to update the plugin to 01.08.04.05.

closed 2024-02-06 by bambu123

arekm, 2024-02-14:
@bambu123 Suggestion: please add network plugin version to "About  Bambu Studio" (unless it is shown somewhere already on running Bambu Studio?)

bambu123, 2024-02-18:
> @bambu123 Suggestion: please add network plugin version to "About Bambu Studio" (unless it is shown somewhere already on running Bambu Studio?)

I think so.