bambulab/BambuStudio #3437: STEP File Translation Poor Quality

CCS86, 2024-01-20:
### Bambu Studio Version

1.8.3

### Where is the application from?

Bambu Lab github releases

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

Imported STEP files are actually a step down in quality from high quality STL.

### Actual results

Here is a comparison between an inserted primitive cylinder, an imported STL and an imported STEP.

[image]

You can see that the STEP import has 44% fewer triangles, and even gets the dimensions wrong.

### Expected results

Here is another example:

[image]

[image]

That same STEP file imported into Onshape looks great:

[image]

Until the slicing engine can be rewritten to run directly on a solid model, it may not be a big deal. It seems that currently the STEP files are being converted into a faceted model, but without any control over the tolerances used.

### Project file & Debug log uploads

[Cylinder.zip](https://github.com/bambulab/BambuStudio/files/13996316/Cylinder.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MklBambu, 2024-01-22:
Could you provide us with the original step file and stl file for analysis?

Thanks!

CCS86, 2024-01-22:
So, this is interesting... the translation issues may be on certain types of faces only. You can see here that the strict cylinder and torus/fillet look much better. But the other faces: cone, b-spline, etc, look really bad:

[image]

Here are these test models:

[test-files.zip](https://github.com/bambulab/BambuStudio/files/14003981/test-files.zip)

MklBambu, 2024-01-22:
Hi, From the file you provided, it can be seen that the size of the step file is much smaller than that of the stl file, so the grid information of the step file is far less than that of the stl file.
[image]
From the grid, it can also be seen that the grid of stl file is finer than that of step file.
[image]
[image]

CCS86, 2024-01-22:
> Hi, From the file you provided, it can be seen that the size of the step file is much smaller than that of the stl file, so the grid information of the step file is far less than that of the stl file. 

STEP files are not faceted, they are solid models, that is why there is a large difference in file size. Despite the smaller size, the STEP file contains more accurate information describing the model geometry.

I'm not sure what program you have opened the step file in, but those facets shown were created by that program.

MklBambu, 2024-01-26:
You're right, the step file is a solid model, but 3D printing requires mesh data, so it needs to convert the solid into a triangular mesh. There are performance and accuracy requirements here. Currently, we are using the triangulation algorithm provided by the OCCT library, which may have some errors.

CCS86, 2024-01-26:
> You're right, the step file is a solid model, but 3D printing requires mesh data, so it needs to convert the solid into a triangular mesh. There are performance and accuracy requirements here. [...]

Hmm, I wonder why STEP file support is touted as a significant feature, when it just converts to a mesh anyway with a buggy library.

LupusTheCanine, 2024-02-01:
> You're right, the step file is a solid model, but 3D printing requires mesh data, so it needs to convert the solid into a triangular mesh. There are performance and accuracy requirements here. [...]

Slicing doesn't inherently require mesh data. What is required is one or more intersection between curve between the model and the slicing plane.

evilDave, 2024-02-08:
I think it very much depends on the resolution/precision of whatever is creating the mash from the model: be that the CAD software, or the slicer... many are finding that their cad software produces better mesh than their slicer, I have found the opposite (step = 132000 triangles vs stl = 54000) with onshape: the mesh from the slicer (i.e. importing a step file) is better quality than the fine mode from onshape (though you can use custom mode and get better results, but the files start to become quite large and the slicers have problems again...)

I for one am waiting for someone to make a move on rewriting a slicer to work directly on the model and not require (or convert to) a mesh... then the quantisation into linear stepper moves can be done much later (at higher resolution, in firmware)

the problem is much more complicated when you don't only have triangles though.

CCS86, 2024-02-08:
> I think it very much depends on the resolution/precision of whatever is creating the mash from the model: be that the CAD software, or the slicer... many are finding that their cad software produces [...]

You are contradicting yourself here. The Onshape presets are obviously too coarse. So, the solution is to fine tune the custom export. Who cares if the file size is bigger? What problems does the slicer have with large STLs?

evilDave, 2024-02-08:
I’m talking about being slow to load, slow when moving models, slicing slower - though I do understand your point and agree that in the end (while the slicing is done on triangles only) then the mesh has to be just as big no matter how it gets into the slicer. I guess my issue here is that I guess I am like most people, I don’t really know what setting I should use to configure the mesh generation and I just want the tools to do the right thing easily. Would you say that the precision (wrt quality of mesh generated) likely goes: your cad software > slicer > onshape - so I see better mesh using slicer but you see worse? I’ll read up what the stl export options mean and try a better stl from onshape and see if anything changes in the rest of the pipeline to print quality. 

I wasn’t disagreeing with the premise of this ticket, that the step file import needs to be configurable (or at least better).

BiscuitsAreForDunking, 2024-03-19:
I hope this gets fixed because I don't remember this being an issue in previous versions of BS and it's made working with smooth curvatures much more of a pain!

tobia, 2024-11-13:
> I have found the opposite with onshape: the mesh from the slicer (i.e. importing a step file) is better quality than the fine mode from onshape

Same with Plasticity. The mesh created by Bambu Studio 1.9 when importing a STEP (when it works) has more uniform polygon size, therefore is more optimized, than what I can get by exporting a STL or OBJ from Plasticity. It is true that other times the STEP import produces obvious artifacts, in which case I need to re-boolean the objects in Plasticity to work around the OCCT import limitations.

I want to add that the new dialog introduced in 1.10 is quite a bit confusing. It's not clear what the default value should be (which was working fine in 1.9) and in fact I can't easily reproduce the imports I was doing with 1.9, which were very good. Except in those cases when the STEP file makes OCCT fail, or in the case of b-spline/nurbs surface poor quality, which is the actual issue being opened here by the OP.

MackBambu, 2024-11-24:
> I want to add that the new dialog introduced in 1.10 is quite a bit confusing. It's not clear what the default value should be (which was working fine in 1.9) and in fact I can't easily reproduce the imports I was doing with 1.9, which were generally great, except in those cases when OCCT generates artifacts, or in the case of b-spline/nurbs surfaces which do have very poor quality. Which was the original issue being raised by OP.

You can think of it as a resolution adjustment option. The smaller the parameter value, the higher the resolution, resulting in smoother surfaces.
https://wiki.bambulab.com/en/software/bambu-studio/step

closed 2025-03-31 by MackBambu