bambulab/BambuStudio #3440: Unsupported OS locale crashes app

hadess, 2024-01-20:
### Bambu Studio Version

git

### Where is the application from?

Built from source by yourself or others

### OS version

Linux

### Printer

N/A

### How to reproduce

Launch `LC_ALL=fr_FR.UTF-8 bambu-studio` on an OS without French locale support (change the locale to another language if necessary).

### Actual results

```
[2024-01-20 11:28:20.831419] [0x00007f29fdaedbc0] [trace]   Initializing StaticPrintConfigs
terminate called after throwing an instance of 'std::runtime_error'
  what():  locale::facet::_S_create_c_locale name not valid
Aborted (core dumped)
```

### Expected results

The app attempts to fallback to a supported locale, such as `C.UTF-8`.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [ ] Project file

hadess, 2024-01-20:
A Flatpak specific work-around is available here:
https://github.com/flathub/com.bambulab.BambuStudio/issues/6#issuecomment-1902062573

hadess, 2024-08-06:
Poke

falkorichter, 2026-10-08:
I have the same Issue!