bambulab/BambuStudio #3443: BambuStudio Blank login screen

iklin, 2024-01-20:
### Bambu Studio Version

01.08.04.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1.Start BambuStudio
2. GoTo Home 
3. Press Login
4. See White Screen

Info: Language Change, try older version, reinstall Plug-in, reinstall BambuStudio doesn't work

### Actual results

[image]

### Expected results

See Login Screen

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/13998677/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

tonycstech, 2024-03-16:
Many pop up windows are blank for me too. including wizard and printer selection

DvoraGames, 2024-06-13:
This also happens in Portuguese.

[image]

### Cause:

This happens because the login URL does not exist in the configured language, and this error only occurs if the language is set in the Language Code Safe function, as it is necessary to translate the web parts (HTML, CSS, and JS). Unfortunately, by doing this, the error arises since the login tries to access a URL that does not exist. I discovered this because once there was a Timeout on this Modal screen, and I managed to see the URL it was trying to access.

[image]

https://github.com/bambulab/BambuStudio/assets/62312359/2cc138c9-527d-45df-8b8e-8311408a6401

StoneLiBambu, 2024-06-28:
Hi @
We will contact the cloud  to configure it and resolve it.

Stone Li
Thanks!

iklin, 2024-06-28:
With  version 1.9.2.57 the login now works for me.

closed 2024-06-28 by iklin

StoneLiBambu, 2024-07-04:
done.