bambulab/BambuStudio #3444: Auto select print profile based on printer selected in device tab, for user with different printers

Photogad, 2024-01-20:
**Is your feature request related to a problem? Please describe.**
It would be convenient if there was an option to automatically select the correct print profile for the printer you are selecting on the device tab, in particular for us users that own both a P1S and X1C printer.

**Describe the solution you'd like**
When selecting the active printer on the "device" tab, the software should automatically select the appropriate printer profile on the "prepare tab", and vice versa.

**Describe alternatives you've considered**
I can do it manually but sometimes I forget to do it.

**Additional context**

[image]
[image]

LJKraus, 2024-01-22:
I would vote against this change.

Currently the Prepare/Preview tabs are independent of the Device tab.  This allows preparing a project while a different project is printing, or even when the printer for the project is turned off.

Photogad, 2024-03-30:
> I would vote against this change.
> 
> Currently the Prepare/Preview tabs are independent of the Device tab. This allows preparing a project while a different project is printing, or even when the printer for the project is turned off.

Easy answer to that, then, is to give users an option to toggle if they want it to work that way or not.

I appreciate your answer about that, and see how that can be a positive thing to be able to do that, but current functionality makes it harder to print everything for us users that have multiple print farms of different Bambu printers. Considering the amount of $$$ spent on 10+ printers it should be considered as an option.