bambulab/BambuStudio #3445: MacOS: no preview window in app

h3lixcoorp, 2024-01-20:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

macOS, Ventura 13.6.3

### Additional system information

CPU: 2,9Ghz i7 Quadcore
Graphic: Radeon Pro 560 4gb
16 gig RAM

### Printer

X1 Carbon

### How to reproduce

Open Bambu studio and choose the stl, then there is no preview or information at all

### Actual results

[image]

### Expected results

I think I should see my 3d preview and print times 🍡 

I think the problem also was posted from someone else

### Project file & Debug log uploads

no crash occurred

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2024-01-22:
@h3lixcoorp 

could you help to upload the logs at /Users/user.name/Library/Application Support/BambuStudio/log

furthermore
is this a 100% reproducable issue? and did it workable before

Haidiye00, 2024-01-22:
Hello, is your GPU an Intel model

h3lixcoorp, 2024-01-23:
Hi again, and thanks for the quick reply. This problem occurred after installing on OSX. Nevertheless, there are the logs: it now works fine
[log.zip](https://github.com/bambulab/BambuStudio/files/14022527/log.zip)

Haidiye00, 2024-01-23:
Hello, can you tell me how you resolved it? Thank you very much

h3lixcoorp, 2024-01-23:
> Hello, can you tell me how you resolved it? Thank you very much

There was basically nothing I did, I searched for problem solving but found nothing except a statement about the openGL feature not being possible in osx, but as I see, after restarting the program it worked. Also maybe to mention: it's not possible to open Bambu twice on Mac, otherwise it'll output the same problem. 

ot: do you know where I can get someone on the line for the shipment of the printer? the EU customer service is very bad and won't reply in time

lanewei120, 2024-01-23:
@h3lixcoorp 
you can create a custom ticket from https://bambulab.com/en/my/support/tickets/create
our CS team will handle your request as soon as possible