bambulab/BambuStudio #3448: This is not very good

hopeinformer, 2024-01-20:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Ubuntu 22.04.3 LTS

### Additional system information

I don't know

### Printer

X1C

### How to reproduce

Load multiple models at once, slice a model... Basically using the program as intended.

### Actual results

It crashes constantly. It's crashed 3 times just today (only a few hours). If I load more than one model its such a pain to arrange. I have to go to PrusaSlicer to arrange my models then export it so the BambuStudio imports the models correctly. If this is based off PursaSlicer then it wasn't done very well. I don't see any evidence of PrusaSlicer in this software. Their software is so much better and easier to use. 

Also, why is it such a pain to get the filament right? If I used slot 4 (in this cases ASA) in the previous slice and update of my model [I should be able to refresh instead of deleting the build plate and loading a "new model" after I make an update to it. Also, it goes back to slot 1 (in this case PLA) and gives me an error the Engineering Plate isn't suitable for PLA. I didn't change it to PLA. I'm using ASA in slot 4. BambuStudio makes it so difficult for everything. Honestly, I wish I could just use PrusaSlicer instead.

### Expected results

Try using PrusaSlicer and go use BambuStudio afterwards. I expect an expensive closed source printer to be better than this, that's the expected results.

### Project file & Debug log uploads

Says it doesn't support the file type. Then what are you asking for?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2024-01-22:
> Says it doesn't support the file type. Then what are you asking for?

You can zip the 3mf file and then upload here

lanewei120, 2024-01-22:
@hopeinformer 

I am sorry our linux version is as not stable as windows version
1. could you show us what steps are you doing before these crashes?
and if you can give some the detailed steps to reproduce it, we are much appreciated;

2. for the auto-arranging
Bambu Studio has an auto-arranging(https://wiki.bambulab.com/en/software/bambu-studio/auto-arranging) for multiple plates slicing, which is better for project-based flow(https://wiki.bambulab.com/en/software/bambu-studio/project-based-workflow)
do you mean your auto-arrange not work?

3. as BambuStudio supports multiple bed types,(https://wiki.bambulab.com/en/x1/manual/compatibility-and-parameter-settings-of-filaments)
there is some logic to check the compatibility