bambulab/BambuStudio #3450: MacOS: Right click on preview '+' button changes layer height making it very difficult to add pause

dclaze, 2024-01-21:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 12.6.6

### Printer

Bambu Lab A1 Mini

### How to reproduce

1. Go Preview tab
2. Move to desired layer
3. Try to right click on '+' button
4. Add Pause

### Actual results

In step 3, it is as you right click the layer, the layer from step 2 gets updated and the pause is added to the wrong layer

### Expected results

Right clicking does not move the layer height

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

amzaldua, 2024-01-23:
Yes, but you can use Control+left click and voilà!

dclaze, 2024-01-24:
@amzaldua can you clarify the work around here? I did CTRL + LEFT (CMD+LEFT on macOS), and I see what looks like a GCODE menu, can you clarify how would I use it to add a pause?

[image]

amzaldua, 2024-01-24:
Ok, in my MacBook a would precede right here.
First, I maintain pulsed the Control button, I move the cursor over the slicer and a made click. Always maintain pushed the con trol button. Then appear a submenu with the pause option. The slice should be positioned in the layer that you can add a pause in the beginning of that.
If you do it directly with the right mouse button, it will happen exactly as you mention: when you press the right button, the bar moves from the layer where you want to insert the pause. It's as you say, they have to make it simpler, but doing it this way, reminiscent of the old school when Macs came with a single button, works fine.
I also had to deal with this matter until I came up with the solution. I apologize if I didn't explain myself correctly at the beginning.

mhwunder, 2024-01-28:
I found using two finger right click on the trackpad to add a layer was impossible to do accurately with my MacBook.  Long ago I started doing Ctrl + click for right click add pause function and never looked back.  I know not a true fix and it shouldn't be as sensative as it is but I wonder if it is just a problem two finger right click?

LiZ-Li-BBL, 2024-02-05:
Thank you for your feedback! We're considering a solution.

stevepopovich, 2024-06-07:
Yep, thanks for the solution with control. With prints with many layers, this was a nightmare. I would love a keyboard shortcut for this. It is something I could look at contributing myself, if you are all open to contributions?