bambulab/BambuStudio #3451: Sync AMS Issue

Aquapura4, 2024-01-21:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac Sonoma 14.2.1

### Additional system information

M2 Chip

### Printer

Bambu A1 Mini

### How to reproduce

Preconditions
1. Have AMS
2. Add filament so that:

Slot 1 = Generic PLA
Slot 2 = Generic PETG
Slot 3 = Generic PLA
Slot 4 = Generic PLA

Start 
1. Click on the filament 'sync' or 'resync'

### Actual results

Software will sync with 5 filaments which is wrong. 

When trying to send to the printer even though all the filaments are the same. The printer will not allow the PETG to be selected. 

### Expected results

Should sync with correct filaments
Should allow selection of PETG when all filaments are the same. 

### Project file & Debug log uploads

[CubeBUG TEST.3mf.zip](https://github.com/bambulab/BambuStudio/files/14000450/CubeBUG.TEST.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Aquapura4, 2024-01-21:
This could also be firmware related because i can't print with PETG when selecting it with the printer either

bam-winterfell, 2024-01-23:
hi@Aquapura4 
The first question : After you click the sync button，external materials will also be synchronized to the prepare page, so the fifth material is external materials,
[image]
[image]

The second question is that the printer will not allow the PETG to be selected. Would you please record a video to recreate the situation when the problem occurred?

bam-winterfell, 2024-02-21:
hi@Aquapura4 
If your problem is not resolved, please create a new issue.

closed 2024-02-21 by bam-winterfell