bambulab/BambuStudio #3453: Segmentation fault in CLI usage using both linux and macOS

NP-nora, 2024-01-21:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 12.7.2

### Additional system information

CPU: 2,7 GHz Dual-Core Intel Core i5
Memory: 8 GB 1867 MHz DDR3
Display card: Intel Iris Graphics 6100 1536 MB

### Printer

Bambu Lab X1 Carbon 0.4 nozzle

### How to reproduce

1. create and run orbStack vm ubuntu 22.04
2. sudo apt-get install wget
3. wget https://github.com/bambulab/BambuStudio/files/13469129/machine.json
4. wget 5. https://github.com/bambulab/BambuStudio/releases/download/v01.08.04.51/BambuStudio_linux_ubuntu_v01.08.04.51_20240117.AppImage 
6. wget https://github.com/bambulab/BambuStudio/files/13469124/process.json
7. wget https://github.com/bambulab/BambuStudio/files/13469135/filament.json
8. wget https://wiki.bambulab.com/knowledge-sharing/stl-files/scraper_grip_by_bambu_lab.stl

9. sudo apt install software-properties-common
10. sudo add-apt-repository universe
11. sudo apt install libfuse2

12. sudo apt-get install libgl1 libegl1 libwebkit2gtk-4.0-37

13. chmod a+x BambuStudio_linux_ubuntu_v01.08.04.51_20240117.AppImage
14. ./BambuStudio_linux_ubuntu_v01.08.04.51_20240117.AppImage --load-settings "machine.json;process.json" --load-filaments "filament.json" --slice 1 --debug 3 --export-3mf output.3mf scraper_grip_by_bambu_lab.stl
15. For macOs 
/Applications/BambuStudio.app/Contents/MacOS/BambuStudio --load-settings "cli_usage_examples/machine.json;cli_usage_examples/process.json" --load-filaments "cli_usage_examples/filament.json" --slice 1 --debug 3 --export-3mf output.3mf  scraper.stl

### Actual results

Segmentation fault. 
[image]

### Expected results

Successfully create sliced file as output, 
as indicated by example command and standard settings provided at https://github.com/bambulab/BambuStudio/wiki/Command-Line-Usage 

### Project file & Debug log uploads

Complete CLI log attached, segfault happens, thus no other output file is produced.
[output_bambu_cli_ubuntu.txt](https://github.com/bambulab/BambuStudio/files/14002614/output_bambu_cli_ubuntu.txt)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2024-01-22:
@NP-nora 

it is caused by "--slice 1"
as there is no 3mf as input, there is no specific plate id can be used as slicing
we will refine this logic later

currently, please use "--slice 0" instead

lanewei120, 2024-04-10:
already included in 1.9 public beta version

closed 2024-04-10 by lanewei120