bambulab/BambuStudio #3454: bambu pla-cf wong temp

dd250032, 2024-01-21:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

When using the default config for bambu lab pla-cf it sets the nozzle temp tp  140 when it should be 240

### Actual results

its clogs the nozzle

### Expected results

i have to manually change temp so it doest clog the setting should have it set for 240

### Project file & Debug log uploads

i don't know how to do logs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-01-22:
Could you provide your gcode files？

LJKraus, 2024-01-22:
Can't reproduce 140° temperature.
UsIng Windows 11,  Studio 1.8.4.51, my PLA-CF defaults to 230°C.  
No clogs on X1C, 0.4 nozzle
[image]