bambulab/BambuStudio #3457: no access sd card bambulab A1

Lefabangoon, 2024-01-22:
### Bambu Studio Version

1-8-4-51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Monterey 12.7.2

### Additional system information

intel core I5 4cor 3,2GHz, RAM 24Go ,AMD radeon R9 M390 2Go

### Printer

BambuLab A1 combo

### How to reproduce

goto "device"
micro SD card
error: initialize failed: not supported on the current printer version

### Actual results

[image]

### Expected results

nothing

### Project file & Debug log uploads

[Casper.3mf.zip](https://github.com/bambulab/BambuStudio/files/14007083/Casper.3mf.zip)
[debug_Mon_Jan_22_07_49_41_959.log.0.zip](https://github.com/bambulab/BambuStudio/files/14007122/debug_Mon_Jan_22_07_49_41_959.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-01-23:
Please wait for new firmware version.

renamed1337, 2025-01-05:
The issue is still present on the newest A1 firmware and Bambu Studio software.

bambu123, 2025-01-06:
Please provide the latest information. Screenshots, logs.

renamed1337, 2025-08-30:
This really should resolved, as a customer i expected so have access to the sd card and webcam in the official slicer. It's crazy this isnt fixed by now.