bambulab/BambuStudio #3465: Display Bambu Studio Version Name In Device Update Page

ThePieMonster, 2024-01-23:
The version number is not present or to difficult to find.

The version of Bambu Studio should be listed here:

[image]

LJKraus, 2024-01-23:
The Studio software is independent of the hardware.

Firmware version is on the Device page because it is a part of the hardware.

The standard location to find the software version of most programs is Help>About
[image]

ThePieMonster, 2024-01-23:
I agree. 
I noticed when going from 1.7 to 1.8 I have a help option under file where as before I did not.

closed 2024-01-23 by ThePieMonster