bambulab/BambuStudio #3466: Studio Crashes When 'Cancelling" Cloud Print Job

ThePieMonster, 2024-01-24:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 64 Bit

### Additional system information

Intel Core i9-10900K

### Printer

X1C

### How to reproduce

When sending a print to the printer select 'cancel' during the cloud sending stage.

[image]

### Actual results

Studio will crash

### Expected results

Studio will not crash

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2024-01-25:
Hi. @ThePieMonster 
Yes, it appears that our main window is blocking, and canceling the print before reaching 70% does indeed cause this issue. We will find a solution for it. Alternatively, you can also cancel the task after it has been successfully sent.

StoneLiBambu, 2024-05-06:
Hi,
This is indeed a blocking function call, but generally not for long.
Cancellation is judged in the middle of a series of function calls. The GUI does block during function calls. But there is a timeout at the end to ensure a non-blocking state.

Thanks!
Stone Li

closed 2024-05-06 by StoneLiBambu