bambulab/BambuStudio #3467: Unable to change flushing volume on Flatpak version of Bambu Studio

AeroBliss, 2024-01-24:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Built from source by yourself or others

### OS version

Linux Mint 21.3 Cinnamon

### Printer

Bambu Lab P1S

### How to reproduce

1. Install Flatpak version of Bambu Studio on Linux Mint
2. Open [this profile](https://makerworld.com/en/models/112380#profileId-129738) inside Bambu Studio
3. Click on "Flushing volumes" under the Filament menu
4. Blank window appears

### Actual results

When running BambuStudio Flatpak
[image]

### Expected results

When running "BambuStudio_linux_ubuntu_v01.08.04.51_20240117.AppImage"
[image]

### Project file & Debug log uploads

Project File: 
[Project.zip](https://github.com/bambulab/BambuStudio/files/14033838/Project.zip)

Log File: 
[Log.zip](https://github.com/bambulab/BambuStudio/files/14033841/Log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

hadess, 2024-01-24:
Known issue, likely a wxWidgets problem:
https://github.com/flathub/com.bambulab.BambuStudio/issues/1

hadess, 2024-01-25:
There will be a test build in https://github.com/flathub/com.bambulab.BambuStudio/pull/12 shortly

AeroBliss, 2024-01-26:
https://github.com/flathub/com.bambulab.BambuStudio/pull/12 fixes issue. Closing.

closed 2024-01-26 by AeroBliss