bambulab/BambuStudio #3468: [ISSUE] with bed leveling, print after firmware update to V01.05.02 on my P1P

Giccen, 2024-01-24:
### Bambu Studio Version

V1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

WIN 10

### Printer

Bambu Lab P1P

### How to reproduce

- I've printed 33X Multiboard grids 8x8 without any issue
- startet the firmware update to from V01.05.01 to V01.05.02 on my P1P
- tried to continue my prints
- all prints around the "full" bed are failing
- the nozzle is hitting the print (with in layer 3) and ripping it up
- **I've now did 4x the FULL calibration of the printer**
- did bed leveling config with each print approach 
**- there is no change**
- it's strange, I hear the nozzle touching the bed due the initial test stripe (L-shape) before starting the real print
- the results are repeatable with other Filament as well

### Actual results

-I've tried to print small objects in the center of my bed
-that works
- it feels like my old Ender 3 :-/

### Expected results

- well it should work like under V1.05.01

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14034928/log.zip)
[8x4 & 8x8 Multiboard.zip](https://github.com/bambulab/BambuStudio/files/14034938/8x4.8x8.Multiboard.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

TCC-up, 2024-01-25:
Hi, are you downgrading over so this doesn't happen? We need more detailed information to locate this issue, can you try to downgrade the printing first and then upgrade the printing (ABA test), and provide the video and this time-end logs uploaded to the work order system via handy, thank you very much for your assistance in troubleshooting this issue~!

redrail1, 2024-06-04:
I'm so mad.  This printer has worked perfectly fine for over a year.  I update and the very next print it starts having these same exact problems.  Is the company hurting for cash?  Did they realize they made too great of a printer so they had to break it themselves?  Its awfully convenient.