bambulab/BambuStudio #3470: Importing from Makerworld to Bambu Studio fails

StargazerVit, 2024-01-24:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: 13th Gen Intel(R) Core(TM) i9-13900H   2.60 GHz
Memory: 64 GB
Display card: NVIDIA RTX 2000 Ada Generation Laptop GPU

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Go it [Model on Makerworld (Just as an example)](https://makerworld.com/en/models/64650#profileId-67854)
2. Click "Open in Bambu Studio"
3. Wait for Bambu Studio to open

### Actual results

After the model seems to be downloaded an error pops up saying "Import to Bambu Studio failed. Please download the file and manually import it."
[image]

### Expected results

Expect the model to be imported in to Bambu Studio.

### Project file & Debug log uploads

Log file: [logs.zip](https://github.com/bambulab/BambuStudio/files/14041004/logs.zip)
No Project file to include as the problem is with importing from Makerworld. Local projects are opening fine.

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2024-02-04:
Hi @StargazerVit 
We have tested the link address, and it appears to be working fine. The tests have passed as well. It could be that your network is causing the download to fail. Please try again using a different hotspot on your phone.

StargazerVit, 2024-02-21:
> Do you have any other questions？

Hi,
I want to retest the flow and understand the root cause of the problem. I
didn't have the chance to test it yet. Will update you ASAP.

Thanks,
Vitali.

closed 2024-04-29 by walterwongbbl