bambulab/BambuStudio #3475: Layer eight's limits are not correctly defined in the Bambu Lab A1 0.2 nozzle preset.

amzaldua, 2024-01-25:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Mac OS Sonoma 14.2.1

### Additional system information

Apple M1 Macbook

### Printer

Bambu Lab A1

### How to reproduce

Go to "Bambu Lab A1 0.2 nozzle preset/Extruder/Layer height limits" and check the MAX and MIN values.

### Actual results

It shows the same values as the Bambu Lab A1 0.4 nozzle:
0.08 for the MIN
0.28 for the MAX

### Expected results

0.04 for the MIN
0.14 for the MAX

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-01-25:
why it should be from 0.04 to 0.14?

amzaldua, 2024-01-25:
Because these values are the same as those defined in all other models of Bambu Lab for a 0.2 nozzle.

QingZhangBambu, 2024-01-25:
@XunZhangBambu check this please

XunZhangBambu, 2024-02-19:
@amzaldua  Thanks. Will fix it soon.

closed 2024-06-17 by XunZhangBambu