bambulab/BambuStudio #3477: Noozle - Spelling Mistake in all BBL machine start gcode

thelandinlook, 2024-01-25:
### Bambu Studio Version

All

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Printer

All

### How to reproduce

A minor spelling error:
In all of the .json printer profile files under /resources/profiles/BBL/machine, "_nozzle_ load line" is misspelled as "_noozle_ load line".
Found under line 63, "machine_start_gcode

Found this while looking to remove the nozzle load line code, and not being able to find it for a while due to searching for the term "nozzle"

### Actual results

Mispelled word
[image]

### Expected results

Not mispelled I guess?

### Project file & Debug log uploads

[Bambu Lab A1 0.2 nozzle.json](https://github.com/bambulab/BambuStudio/files/14051421/Bambu.Lab.A1.0.2.nozzle.json)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XunZhangBambu, 2024-02-04:
Thanks for your patience in finding the spelling errors ,we will correct it soon

jwise, 2024-05-15:
I remember this one from #617!

But I have to say that -- as the official position of the X1Plus team -- I miss the noozle already.  We always thought that typo was charming.

closed 2024-06-17 by XunZhangBambu