bambulab/BambuStudio #3479: Can't print TPU with AMS connected

hopeinformer, 2024-01-25:
### Bambu Studio Version

1.7.6.92

### Where is the application from?

Bambu Lab Official website

### OS version

Linux Ubuntu 22.04.3 LTS

### Additional system information

Not sure why this matters
Dell Inspiron 5559
Intel core i7
2TB
Ubuntu 22.04.3 64-bit**

### Printer

X1C

### How to reproduce

Have the AMS connected and load TPU from the back. *Not loading the TPU from the AMS. Select the 5th slot with the TPU in BambuStudio. Slice the model with TPU settings and click "Print Plate." Bambu Studio gives error and won't print. I have to unplug the AMS, click print again, then plug the AMS back in again.

### Actual results

 See comment above in "How to reproduce." 

### Expected results

I shouldn't have to disconnect the AMS every time just to print from the loaded filament. It's already setup, it just has to print it. But instead I have to unplug the AMS unit every time.

### Project file & Debug log uploads

There's not a file. It's the printer not the file.

### Checklist of files to include

- [ ] Log file
- [X] Project file

robere2, 2024-01-25:
Just confirming, did you uncheck the "Enable AMS" box in the "Print Plate" dialog?

keithel, 2024-01-26:
I cannot reproduce your issue. I've been able to print TPU outside of the AMS -- the important thing to pick is to uncheck "Enable AMS" in the print confirmation dialog, as @robere2 mentions. If you do this, it should show a dash under the filament slot/type/swatch widget, and no request to pick a slot. I'll show what's seen unchecking or checking the "Enable AMS" checkbox:
[image]
[image]
You want to uncheck "Enable AMS", then all should work.

walterwongbbl, 2024-01-26:
@robere2 @keithel 
Thank you very much for your answers

keithel, 2024-01-26:
@walterwongbbl it seems to me that the printer should know if there is an external spool in use by detecting that there's filament in the hot end (if the user loads it first from the printer). In that case, I would think there could be a way for Bambu Studio to bring up the print plate dialog without the "Enable AMS" checkbox checked, and this confusion would be minimized to only happen when no filament (that isn't from the AMS) is loaded.

walterwongbbl, 2024-01-29:
@keithel 
You are correct, but some users have already developed mature usage habits, and if we change the current interaction, some users would prefer us to revert back.

closed 2024-02-18 by walterwongbbl