bambulab/BambuStudio #3480: App crash during painting mode

darkneon2002, 2024-01-25:
### Bambu Studio Version

1.8.4..51

### Where is the application from?

Bambu Lab Official website

### OS version

win 11 x64

### Additional system information

i7-14700KF, RTX2080Ti 11GB, 64GB DDR5

### Printer

P1S

### How to reproduce

import multiple models on a single plate
select 1 object
open painting mode
SHIFT + Left Mouse on the object list that "should" select all or multiple objects

https://github.com/bambulab/BambuStudio/assets/10987492/46007720-8274-473d-be2c-728072d7a698

### Actual results

Bambu Studio will hard crash

### Expected results

A message that selecting multiple models during painting is not possible or
it will work (would not expect this) or
nothing if it's not possible

### Project file & Debug log uploads

[paint to crash.zip](https://github.com/bambulab/BambuStudio/files/14055555/paint.to.crash.zip)
[crash_Thu_Jan_25_16_42_39_0.zip](https://github.com/bambulab/BambuStudio/files/14055572/crash_Thu_Jan_25_16_42_39_0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-01-30:
@darkneon2002  Is this bug reproducible stably, or does it only appear this once?

darkneon2002, 2024-01-30:
yes very stable bug, I reproduced it on various projects with multiple objects.

darkneon2002, 2024-01-30:
very stable and reproducible

https://github.com/bambulab/BambuStudio/assets/10987492/584720f9-b8ae-4c17-986c-266fc5513f3e

zhimin-zeng-bambulab, 2024-01-31:
We do not support selecting multiple objects for painting at the same time. We will limit this behavior and resolve the crash caused by this operation in the next version.

Haidiye00, 2024-03-01:
This issue has been fixed. Multiple model_object selections have been disabled in the Paint mode. Please wait for version v1.9. Thank you

zhimin-zeng-bambulab, 2024-03-29:
It has been fixed in v01.09.00.60, you can have a try

closed 2024-03-29 by zhimin-zeng-bambulab