bambulab/BambuStudio #3481: Bambu Studio 'forgets' lan mode access if printer is not online when print is startedd0817b8a

DaafSamson, 2024-01-26:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz   2.21 GHz / 32,0 GB Ram

### Printer

X1C

### How to reproduce

The following assumes you are using the printer in LAN Only mode.

1. turn off the printer
2. start a print in studio
3. wait for connect to the printer to time out
4. turn on the printer
5. reconnect in the select printer dialog
6. get prompted for the printer access code again.

### Actual results

Connection fails, with error: "Connect 3DP-XXX-XXX failed! [SN:XXXXXXXXXXXXXX, code=-1]"
[image]

Reselecting the printer after it's been turned on, prompts for access code
[image]

It seems that on the timeout the access code is deleted from the system.

### Expected results

Expected result would be that after turning on the printer, the software still has the lan mode access code an can simply connect

### Project file & Debug log uploads

unrelated to project. happens on all projects

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DaafSamson, 2024-01-26:
I could not easily find related code. Perhaps this is part of the network plugin, not studio itself?

LJKraus, 2024-01-26:
Try using the current version 1.8.4, which includes a [connection timeout bugfix.](https://github.com/bambulab/BambuStudio/releases#:~:text=Fix%20the%20issue%20of%20frequent%20printer%20connection%20timeout%20%233045)

walterwongbbl, 2024-01-29:
@DaafSamson 
Because we do not manage the access code settings, we hope that users can re-enter them every time. Although it may be a bit troublesome, there is currently no better way. We are conducting large-scale optimization, including input of access code. Thank you very much for your feedback.

closed 2024-02-21 by walterwongbbl

AlphaEndgame, 2024-07-21:
This is possible to achieve on the latest version 1.9.3.50, at least on Windows:

Open Bambu Studio, go to Device, No Printer, and select your LAN only printer. Input its access code and make sure you can connect to it. Afterwards, close Bambu Studio.

Navigate to the following file:

"C:\Users\<User>\AppData\Roaming\BambuStudio\BambuStudio.conf"

Edit this file and search for "user_access_code"

You will find something like this:

    "user_access_code": {
        "<PRINTER_ID>": "<ACCESS_CODE>"
    }

Now, change the name of the field to just "access_code", like this:

    "access_code": {
        "<PRINTER_ID>": "<ACCESS_CODE>"
    }

Save the file. Now, Bambu Studio will remember your printer's access code each time it boots up.

FlorianKuehne, 2024-08-04:
It Works in Linux too. Thanks a lot.

diver-ch, 2024-08-14:
@AlphaEndgame:
Thanks alot! This was buggin' me since several weeks now. In my case there where 2 entries, "access_code" and "user_access_code". The "access_code" contained a old one, the "user_access_code" was correct. I've just updated the "access_code" and just left the user entry aswell > so far it works fine.

Rello, 2024-08-14:
macOS:
user/Library/Application Support/BambuStudio/BambuStudio.conf

oltsukoo, 2024-08-19:
This also works on OrcaSlicer, just make the same change in C:\Users\<User>\AppData\Roaming\OrcaSlicer\OrcaSlicer.conf.

woolmonkey, 2024-10-26:
OMG thanks so much this was so annoying and it seems to have fixed it.  My thought they are making it less convient for LAN mode since all these outages more people are probably using it and they don't like it.

Gruensil, 2024-11-13:
Solved my problem on Linux too. I installed with flatpak and config file was under ~/.var/app/com.bambulab.BambuStudio/config/BambuStudio/BambuStudio.conf

stilsch, 2026-01-06:
Changing `user_access_code` to `access_code` in `~/.var/app/com.bambulab.BambuStudio/config/BambuStudio/BambuStudio.conf `did **not** solve the issue for me.

LJKraus, 2026-01-06:
I keep Studio on my PC, mainly to see what others are complaining about.  I use OrcaSlicer to actually print.

I made the "user_access_code to access_code" change long ago, but I see that it has been overwritten and changed back to user_access_code at some point.  The correct code _**is**_ saved in BambuStudio.conf.  

In LAN mode, the printer has been "forgetten" every time I start Studio and sometimes even during long sessions. The Device tab shows "No Printer" until I hit the "+" and select the printer. I don't need to re-input the IP or access code, but I can't see the printer status or send a print until I've done the No Printer dance.

In contrast, the printer is **always** present in OrcaSlicer, no need to re-enter codes or IP, no need to re-select, it's just always available. 

This makes the situation with Studio all the more ridiculous. OrcaSlicer solved the problem nearly a year ago, and Bambu cannot (or will not?) take advantage of the available open source solution to fix this aggravation for their Studio users.