bambulab/BambuStudio #3482: Periodically (regularly) crashes

JBFUK, 2024-01-26:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Processor	Intel(R) Core(TM) i5-4570 CPU @ 3.20GHz   3.20 GHz
Installed RAM	8.00 GB
System type	64-bit operating system, x64-based processor

### Printer

Bambulab X1C

### How to reproduce

Open, setup plate, slice, start print.  Leave app open while printing.  Later come back and try to switch back to the prepare tab or start a new project.

### Actual results

App hangs and then closes.

### Expected results

App switches to prepare tab or starts new project.

### Project file & Debug log uploads

[Fri_Jan_26_00_24_45.log.zip](https://github.com/bambulab/BambuStudio/files/14063477/Fri_Jan_26_00_24_45.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

JBFUK, 2024-01-26:
Sorry for lack of project file, I don't have one, but this happens often so it's not realted to one specific project/plate.

JBFUK, 2024-01-28:
Project File Attached
[1x_spindle_male_driven_Pastamatic_Filament_Winder_by_GekoPrime_2023-06-03.zip](https://github.com/bambulab/BambuStudio/files/14077834/1x_spindle_male_driven_Pastamatic_Filament_Winder_by_GekoPrime_2023-06-03.zip)

bambu123, 2024-01-29:
It crashed in C:\Windows\System32\ig75icd64.dll.

To analyze DMP files, we need you to provide the following DLLs on your system:

C:\Windows\System32\ig75icd64.dll
C:\Windows\System32\KERNELBASE.dll
C:\Windows\System32\msvcp140.dll

JBFUK, 2024-02-03:
[BambuStudioCrashDLL.zip](https://github.com/bambulab/BambuStudio/files/14151068/BambuStudioCrashDLL.zip)

Files attached.

bambu123, 2024-02-06:
It crashed at 'Slic3r::GUI::GLTexture::render_sub_texture'.

lanewei120, 2024-02-18:
@JBFUK 
sorry for the late reply

it is related with opengl rendering, could you help to do following things:
1. send more dmp file to us, we need to confirm whether they are crashed in the same stack or not
2. is the opengl driver on your system modified before? have a try for different ig75icd64.dll from https://www.dllme.com/dll/files/ig75icd64