bambulab/BambuStudio #3483: Hot end temperature error

RegT10, 2024-01-26:
### Bambu Studio Version

18451

### Where is the application from?

Bambu Lab Official website

### OS version

OS X latest version

### Printer

Bambu A1

### How to reproduce

print an object in PETG basic with a 0,6mm nozzel

### Actual results

hot end error during first final top lay due to fan running at 100% and over cooling the hot end

### Expected results

If I reduce fan speed to 40% or less the problem is resolved

### Project file & Debug log uploads

no dubug files 

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-01-29:
Could you provide 3mf and gcode files?

RegT10, 2024-01-29:
sorry no, this happened on all the parts made using a 06mm nozzle and strength setting in studio lab, but I didn’t keep the files.

QingZhangBambu, 2024-02-04:
which filament preset did you choose? Generic PETG or Bambu PETH basic?

RegT10, 2024-02-04:
Bambu

Xiuqi-Huang-CN, 2024-02-05:
@RegT10 
[image]
Hello, 
    RegT10. I'm here to give you some advise, and hope that I can provide any help.
    First, in my experience, among common filaments, only PLA and TPU needs very high fan speeds to ensure enough cooling, while PETG, ABS, ASA, PC, PA, PET-CF/GF, PA-CF/GF, PATH-CF and other high printing temp filaments is different, and they need less fan percentage in most areas of the prints to ensure enough cooling and good qualities, and too high fan speeds could lead to warping, poor interlayer adhesion and other issues (that‘s why only 40% fan is set as the maximun for most areas in Bambu PETG Basic) . So, please do some tests, and then set as less fan percentage as possible when print PETG if the prints‘ cooling is enough.
   Secondly, during the printing, when the actuall nozzle temperature is much lower than the set one,  errors and warnings will show up. And this is a kind of protective measures to avoid some possible issues. When the fan speeds are too high, and the nozzle‘s silicone sleeve is broken or missing and can not peserve heat, the nozzle temperature will decrease and be much lower. So, you may need to check whether the nozzle‘s silicone sleeve is in good condition or placed well or not. If necessary, you may need to reinstall it or replace it with a new one.
   Hope that you can get any help. If you have any further question or trouble, please contact us.

RegT10, 2024-02-05:
Thanks for your help, the silicone sleeve was damaged and after replacing as you said the problem was no longer a problem.
Many thanks
Reg

Xiuqi-Huang-CN, 2024-02-06:
@RegT10 Happy to know that the problem was solved.

closed 2024-02-06 by QingZhangBambu