bambulab/BambuStudio #3485: Add ability to choose color when using the Text tool

nowozin, 2024-01-26:
**Is your feature request related to a problem? Please describe.**
Bambu Studio has a Text tool which enables the easy addition of text to a surface.  The color of the text is always the same as the color of the object.  If I want to add text with a _different_ color than the underlying object, this is currently a difficult two-step process: 1. add the text, and then 2. manually change the color of all the text faces on the object.

**Describe the solution you'd like**
Add a color selector to the Text tool which allows one to choose the color of the text _before_ adding the text.  The color choices should be from the available AMS materials currently setup, just like in the Coloring tool.

**Describe alternatives you've considered**
The alternative is the current method of manually changing all text faces, which quickly becomes time-consuming for longer texts.

**Additional context**
The attached screenshot shows the 1.08.04.51 version Text tool dialogue with no color choice available.  As an example of a difficult operation in Bambu Studio, how would I add a red text on top of the flat surface?

[image]

RicVilla78, 2024-01-30:
When you create a text it creates a suboject directly which you can assign another color for the full text without painting.
[image]

MadMax389, 2024-01-31:
I agree with the OP. It would eliminate the steps required to change the text color. Sometimes it's hard to see the text if the object and text are the same color. I would also like to see the text dialog retain the last settings. This would make multi-line text much easier to implement since you have to reset font, size, depth etc every time you add text.

eingemaischt, 2024-02-01:
Furthermore: Currently it is not allowed to add a text with a thickness of 0 - so a text without any embossing, only in another color....

zhimin-zeng-bambulab, 2024-02-02:
@eingemaischt  If you want to add text with a thickness of 0, you can try changing the text part type to a modifier type.

nowozin, 2024-03-18:
@RicVilla78 @zhimin-zeng-bambulab thanks, this helped and allowed me to set the color in a two-step process; however, setting the color using this separate dialogue is not easily discoverable as a user.  Allowing the user to set the color when creating the text is a simpler mental model (and adding colored text is a great use case for Bambu Lab + AMS printers).

@eingemaischt +1, setting the thickness to zero right when creating the text would be really useful in my case.