bambulab/BambuStudio #3486: [Issue] Bambu Studio crashes when trying to view video from printer.  Debian 12

BambuLabDiscord, 2024-01-26:
Every time I try to view the video from the printer it crashes or freezes.  BambuStudio_linux_ubuntu_v01.08.04.51_20240117

OS: Debian GNU/Linux 12 (bookworm) x86_64
Kernel: 6.1.0-17-amd64
DE: Plasma 5.27.5
CPU: Intel i7-5820K (12) @ 3.600GHz 
GPU: NVIDIA GeForce GTX 980 

Tried to run: $ gdb Bambu_Studio_linux_ubuntu_vXXX.AppImage as per the last comment https://github.com/bambulab/BambuStudio/issues/3223.  When I run bt I get 
#0  0x00007ffff0536e6e in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#1  0xe44fa85b8de13e4e in ?? ()
Backtrace stopped: Cannot access memory at address 0xe44fa85b8de13e4e

bambu123, 2024-01-29:
Maybe the same as https://github.com/bambulab/BambuStudio/issues/1814
and https://github.com/bambulab/BambuStudio/issues/1665#issuecomment-1550380137

could you have a check in the above issues?

belzebub40k, 2024-02-20:
Ran into the same problem with Debian 12. Using the Fedora AppImage works for me.

voglster, 2024-05-24:
This is still an issue with the debian appImage

meo209, 2024-09-07:
Got the same problem on Arch Linux running hyprland.

bambu123, 2024-09-09:
We are going to release the player refactoring in the next version.