bambulab/BambuStudio #3488: Drag to rearrange filament location

ironslaves, 2024-01-27:
**Is your feature request related to a problem? Please describe.**
It is a pain to re arrange filament slots to match painted 3mf projects downloaded from maker world. 
**Describe the solution you'd like**
It would help a lot if I could rearrange the slot number that the filaments are located at so the painted colors match whats loaded by clicking and dragging the filament color boxes around in the filament section.

MadMax389, 2024-01-31:
Can't you do this already by assigning whatever colors you have in your AMS to whatever slot you want in the slicer?
[image]

krapclap, 2024-02-02:
No, I think I should describe it better. 
When you load a profile, it loads the profile creators AMS setup as well. I then have to resync my ams and the colors get changed on the model. The logical solution for this is to re load the filaments into the slots already assigned and then sync the AMS or to repaint the colors.  By dragging the color boxes around you don't relocate the filament, but instead you can quickly repaint the object(s) by switching paint colors on the object.  So if 1 is blue and 2 is red, Dragging the little color boxes over each other will switch them So anything painted blue is now painted red and anything painted red is now painted blue.

MadMax389, 2024-02-02:
Perhaps I should describe a little better. I am assuming the problem is that when you download a MW model, the slot colors don't match what you have loaded in your AMS, or you may not have one or more of the colors specified in the profile. This happens to me all the time. My point is **DON'T RESYNC**, simply assign the colors from your AMS into the correct slot. The slot numbers in the slicer do not need to match the slot numbers on your AMS. You can either match or change colors this way.

For example, assume this image is a model I downloaded from MW with three colors, white red and blue in slots 1, 2 and 3. The cube is white and the circles are painted. I have 8 slots in my AMS and I want to print it in different colors with different filament profiles.
[image]

At this point, don't resync because all the slicer slot colors will change. Now I want the cube to be purple, and the circles to be yellow and black. DON'T RESYNC. Click the filament dropdown box and pick the colors that you want the cube and circles to be from your AMS. This will change the color and the filament profile to whatever you have loaded in your AMS.

[image]

Finish assigning all the colors, slice and print. You can see that slicer slot 1 is mapped to AMS slot B1, slot 2 = A4 and slot 3 = A2.
[image]

krapclap, 2024-02-02:
I see, I thought each color had to be tied to the same AMS slot as the number on the color. I had a fundamental misunderstanding of how it works. I assumed that moving the filament with that menu meant I physically relocated the filament to the number on the color. 

now I have to ask. Why are there numbers on the colors. 

Also, thanks for the clarification.

MadMax389, 2024-02-02:
The numbers on the colors in the slicer are just placeholders for the filaments used in the model. Once I realized these slots don't correlate to AMS slots, it changed my workflow. Now, when I create a new model, especially for Makerworld, I only add the number of filaments I'm going to use in the model using the little + and - symbols on the filament dialog. If I sync first and then wind up using slot 7 to paint something, the project will require at least 7 slots, even though I may only be using 3 colors from my AMS. My new rule of thumb is "Never Sync". It makes it a lot easier when I switch between my 4 printers where my AMS's may not have the same color arrangement.