bambulab/BambuStudio #3493: X1C Calibration Routine Directly Results in Oozing + Blobs on First Layer

coreyward, 2024-01-27:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS

### Additional system information

Unrelated

### Printer

X1C

### How to reproduce

Slice and print with PETG. Enable flow calibration and bed leveling when prompted.

### Actual results

The printer prints a series of lines during flow calibration, and then the print head sits hovering over the bed with the nozzle at temperature for around 20 seconds while the print is checked. During this time, PETG oozes out of the nozzle and builds up on the print head and is dragged about. Then, once the nozzle is thoroughly covered in filament, it starts the print. This results in the print needing to be stopped nearly 100% of the time.

### Expected results

After printing the flow calibration lines, the printer should retract the filament and reduce the temperature of the nozzle. Once the nozzle is unlikely to be oozing, the printed lines can be checked. Finally, the printer should perform a purge and/or wipe before the priming line (rear-right quadrant lines) before starting the print.

### Project file & Debug log uploads

Not applicable.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

coreyward, 2024-01-28:
It turns out a similar thing happens even when not running bed leveling or flow dynamics calibration. The prime line at the front of the build plate is laid down and the nozzle is clean, but then the print head hovers a few mm above the print surface for around 5 seconds, during which a blob starts to accumulate on the nozzle. The printer then starts moving again to print the rear prime lines and start printing, now with a big blob on the nozzle. 

I'm not sure if this routine is something that can be modified at all with GCode, but it is the source of almost all of my frustration with the X1C. It seems like it's a printer that is sold as one that can print almost any material, but this behavior makes it unsuitable for anything other than PLA. I get better results with PETG on the A1 Mini and Prusa MK4, so I'll likely exclude the X1C from further expansions unless this behavior is improved.

dMopp, 2024-03-25:
Same here, but heavy depending on the Filament. JAYO PETG is okay (still a blob but small enough) but "dasfilament" have a big blob. SILK PLA has also a bigger blob but here its mostly fine as well. It would be better, if the head would move during the first "blob" so it not starting while standing still... (or just do a small circle across the initial blob, so we avoid to hit that blob again)

gaolegao-lx, 2024-08-07:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

coreyward, 2024-08-09:
@QingZhangBambu It looks like @wineBambu hasn't had a chance to take a look at this in the last 5 months. Is there someone else it should be assigned to?

KrickLogan, 2024-11-06:
I have tried tweaking countless filament and print settings in an attempt to resolve PETG buildup on the nozzle. I was never successful. After closely watching the printer during the print preparation and first layer I realized that the issue described by @coreyward is the primary cause.

dMopp, 2024-11-06:
I was able to solve it. I will add my changes to the printer start-code later here in the thread (I use it for all filaments)

MartinChan3, 2024-11-06:
It's a good way, I will try it with my P1P like @dMopp and Bambu PETG Basic.

molotovgod, 2024-12-24:
I have a similar issue here @dMopp the same thing is happening after the nozzle cleans itself before a print, can I use your line of GCode for that?

In other words, after calibration happens, bed leveling, flow calibration, etc., and the nozzle is cleaned before printing. After purging the filament and discarding the waste, the nozzle (in my case) starts with a larger load of filament for the initial deployed line. This is obvious because it starts building up on the nozzle, it creates defects or entirely ruins the print if not removed.

Thanks

dMopp, 2024-12-26:
Hey, if its curling up, you might have a partial clog. But anyway, my adapted / modified calibration line might help, because it’s stopping extruding a bit earlier, it’s worth to try it.